forkineye/ESPixelStick #130: Flash wear on reboots

atoudam, 2018-10-09:
Hi

if i understand this correctly the current release maybe susceptible to flash memory wear if a device was to go into a reboot cycle (aka if your WiFi AP is inaccessible and you don't have soft AP on)
based on this info
https://github.com/esp8266/Arduino/issues/1054
i think the line
WiFi.persistent(false);
should be added before all calls to wifi.begin() and wifi.disconnect() 
possible to be safe also before wifi.mode()

penfold42, 2018-10-10:
I think you’re right - maybe also just remove the hard coded credentials too - does anyone use them ?
During development I pre-fill config.json

forkineye, 2018-10-10:
Good catch, wasn't aware of that behavior.   Should only need to be added once since we don't ever sleep.  I'll add it on the next commit.  I don't think the hard coded credentials are hurting anything and I know some people use them.  They're not really needed though since everything is in the json config now.

atoudam, 2018-10-12:
happy to help, personally i like the hard coded "back up" credentials, I keep the the show network info in the config and my development network hard coded , this way i can move between the two when the show network is down.

penfold42, 2018-10-15:
Just looking they the code - should we refactor it to take advantage of the ESP’s auto connection on startup ?

If it’s connected and in station mode just jump to IP configuration

forkineye, 2020-12-01:
@MartinMueller2003 From looking at the rework you did on wifi management, this shouldn't be an issue in 4x, corrrect?

MartinMueller2003, 2020-12-01:
That is correct. We can close the issue.
Martin

closed 2020-12-01 by forkineye