forkineye/ESPixelStick #131: FW upload, to much data

atoudam, 2018-10-09:
Hi,
the current method of uploading firmware i believe has a small issue where some browsers after they have posted all byte packets of the file (and have received the ws.'X6' back etc) will still continue to send additional  byte packets from the file (starting from index 0 again ) In my testing roughly the first 130kb of the file.
this only happens on occasion and not if i do a post via node.js/requests.
i believe this is due to ESPixelStick not sending a "correct" response to the updatefw post.

the solution would be to change the /updatefw response from
ws.textAll("X6");
to 
request->send(204);ws.textAll("X6");

full function
    // Firmware upload handler
    web.on("/updatefw", HTTP_POST, [](AsyncWebServerRequest *request) {
         request->send(204);ws.textAll("X6");
    }, handle_fw_upload);

note, this has not likely been noticed/an issue with the current code as the fw upload triggers a reset and so those additional packets have nowhere to go, also because of efupdate.process(data, len) its likely if some of those packets got in before the reset they would be disregarded. I only found this as I am experimenting with using similar upload code to upload files to the ESP without doing a reset after.
 
testing on Firefox OSX

penfold42, 2018-10-12:
I noticed the missing response while troubleshooting something else

I think it needs to go here :
Line 369/370:
https://github.com/penfold42/ESPixelBoard/commit/589c1b941c66a70791cd5d161e50599d77b7e503

penfold42, 2018-10-15:
#132 includes a fix

penfold42, 2018-10-20:
This has been merged - can you please test ?

penfold42, 2018-11-02:
Close ?

atoudam, 2018-11-07:
yes, However I personally haven't tested this version, though i have a similar fix in my working clone.
the serial fix will work 95% of the time however crashes can still happen with serial data coming in. I'm starting to narrow down the cause , though i have some of my own hardware that is also causing serial based crashes.

penfold42, 2018-11-07:
My serial testing wasn’t exhaustive if I’m honest.

It was done in ws2812 mode, 115200 baud and pasting many kb of text via putty+screen with a (probably clone) ftdi based USB serial adaptor.

It survived all that which was a million times better than before !

atoudam, 2018-11-07:
I would expect that to be good enough testing, the only reason I see crashes is i have a serial device connected full time to my dev unit. the DMX mode crash (in my other issue) is a bigger problem which i have solved (if you agree with the modifications)