forkineye/ESPixelStick #134: Using firmware on NodeMCU with GS8208

SJHOne, 2018-10-20:
Using the Arduino IDE I was able to get the ESPS firmware and data uploaded to a 'bargain' NodeMCU I bought from Banggood. 

I bought some 12v light strips with GS8208 LEDs in them. 

I am unable to get this configuration to drive the strips. I fed the output of Pin2 (4) through a 74HCT08 AND gate to raise to 5v, then pass this signal into both DD1 and DD2 of the light strips.

After an initial all LED white flash, nothing happens. Selecting test patterns through the web UI does not help.

I scoped the output and I noticed that the signal rate appeared to be closer to 746kHz than 800kHz - does anyone know if this is too much of a drift?

I am thinking that maybe I need to adjust the timing constants to get the signals closer to 800kHz on these boards.

Thanks,
Steve

forkineye, 2018-11-27:
Are you running at 160Mhz?

sauloonze, 2019-02-26:
@SJHOne I have a GS8208 led strip here and I have a similar issue as you. I am having a problem to find out how to raise the output signal to 5v. Can you please give a hit on it? Do you have a diagram?