forkineye/ESPixelStick #135: esptool Update

eps000, 2018-10-30:
I don't know if anyone else had this issue so I wanted to share what I found. I successfully flashed 15 out of 20 esp-01 but had issues with the remaining 5. I tried to flash them using the stable v3 firmware and the travis version on Mac,Windows and Linux with no luck. The issue I kept getting was: 

warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01 0x04
error: espcomm_upload_mem failed

After many attempts I was finally able to flash the firmware by replacing the esptool thats located in the bin/operating system folders with an update esptool 0.4.13. I downloaded it from https://github.com/igrr/esptool-ck/releases/tag/0.4.13
After doing this it worked perfectly every time. I don't know if others have had this issue but I hope this helps.

jcollie, 2018-12-25:
I wasn't able to get the upload to work on my D1 mini v3.1.0 with either the included version of esptool or the version mentioned above. I needed to use esptool.py v2.5.0 with this command:

esptool --chip esp8266 --port /dev/ttyUSB0  --baud 115200 write_flash 0x000000 firmware/serial-travis.bin 0xDB000 firmware/spiffs.bin

forkineye, 2019-01-06:
First I've heard of issues with esptool, however I just updated it so 0.4.13.

closed 2019-01-06 by forkineye