forkineye/ESPixelStick #137: DMX Mode , near instant crash on Serial RX

atoudam, 2018-11-07:
Hi

there are issues on Serial RX which i have been looking into, one of the worse (and easiest to reproduce) is when using DMX mode (and receiving serial), I would often get a Exception 9 error, unaligned address error caused by the function getFifoLength()

this I believe is due to the method the DMX mode is using to get the number of bytes waiting in the TX FIFO of SEROUT_UART, I don't fully understand how this function works in its current form, however making the below change (which is how I understand this should be done based on the ESP spec etc) fixes all crashes.

static inline uint8_t getFifoLength() {
        return (U1S >> USTXC) & 0xff;
    }

this of course hard codes the serial output that DMX output will use however that seams like a good trade off.

if this change is to be made then there are other parts that can be cleaned up ie

    static inline void enqueue(uint8_t byte) {
    // this was working (aw)    ESP8266_REG((0xF00*SEROUT_UART)) = byte;
     U1F = byte;
    }

thoughts? happy for someone with more experience in this area to put through a pull request, or  I can do it?

penfold42, 2019-01-16:
I can reproduce this - even with my recent workaround in place.

I wonder if this is related to https://github.com/esp8266/Arduino/issues/5555

forkineye, 2019-01-16:
@penfold42 I would agree.  I'll look into it further.

MartinMueller2003, 2020-11-20:
This is no longer an issue on the Unify branch

closed 2020-11-20 by forkineye