forkineye/ESPixelStick #138: Feature Request: Hardware toggle of test mode

Jchuchla, 2018-11-13:
It would be very handy to be able to enable/disable, or toggle thru the test modes on the ESPixelStick via a hardware push button.  
Perhaps the existing  program button could be used for this?  Or perhaps the clock pin on the pixel connector could be set up for this functionality.

Woobatron, 2018-12-19:
Second!

penfold42, 2019-01-06:
I’ve kinda got this in my espixelboard fork.
I have a rotary encoder to select colours in RGB or HSV Color space.

If forkineye isn’t interested, open an issue on my fork and I’ll look into it one day

forkineye, 2019-01-06:
For ESPixelStick hardware, this would have to be the prog / flash button.  Maybe a long press to enter / exit the mode then short presses to toggle between a preset effect list - like static, rainbow, chase.

Jchuchla, 2019-01-06:
That would be more than acceptable.  Originally I was thinking maybe we could leverage that clock pin on the pixel header and wire in a switch or jumper.  But a button that’s already there is even better.

forkineye, 2019-01-06:
On the current hardware, they're actually the same (GPIO0).  That'll be changing in the next hardware release though as clock will actually be tied to SPI so we can do clocked pixels.

MartinMueller2003, 2020-12-03:
Prototype code is available in the MartinMueller2003/ESPixelStick fork.

MartinMueller2003, 2020-12-15:
This code has been pushed to the Unify branch.

closed 2020-12-21 by forkineye