forkineye/ESPixelStick #139: Flickering Issue on NodeMCU 1.0 and Wemos D1 Mini

redwolf3, 2018-11-27:
I'm attempting to use a couple of NodeMCU 1.0's or Wemos D1 Mini's running the ESPixelStick firmware to control about 500 12 RGB LED's (WS2811). 

Everything seemed to be fine when running basic tests in the lab (garage), but now that I'm setting up my show I'm running into issues with my pixels flickering on certain colors.

After a bunch of experimentation it seems like the issue is only occurring when I have the brightness of individual channels set to anything other than near max brightness. It doesn't seem to matter what color or mix I am using, if I have the brightness of any channel set below about 250, I get flickering until I get down to about 50 (at which point you really can't see the light very well).

I've read through a lot of common issues and have tried the following things:

1. Initially I was running the latest version which looks like it has some prototype changes so I went back to Release 3.0. This had no affect.
2. I verified that I am programming the ESP8266's with 160 MHz.
3. I tried disabling the Wifi Status Light using `wifi_status_led_uninstall()`. This had no affect on the flickering (it also didn't seem to turn the WiFi status light off, it just disabled the blinking).
4. I've tried both my NodeMCU 1.0's and D1 Minis and they both exhibit the exact same behavior.

Additional information:
- The cable run from my MCU's and the First LED is less than 1 foot
- I DO have a 330 ohm resistor on the signal line between the MCU and the first LED
- If I light all of the pixels at the same color they all flicker at the same rate or close to the same rate
- I get the same behavior whether using the built-in test utilities or driving the colors over e1.31 (from Xlights in Test Mode)
- My issue seems to match the description and symptoms under https://github.com/forkineye/ESPixelStick/issues/26, except that I DO have a logic level converter that is confirmed working (based on tests with some other firmware)

Looking at the output of the LED's, it ALMOST looks like there is some sort of timing drift that is causing them to get inconsistent signal timing. I suspect that max brightness works fine since this is effectively all "1" values, where as the drift affects the signal more at intermediate levels, but this is just a guess.

I also have a very simple driver / e1.31 adapter I wrote before I found the ESPixelStick firmware. This driver is written using a synchronous e1.31 library and the NeoPixelBus library. This driver does not exhibit the flickering at any color level or range I have tested with. I have also successfully run a couple of sample shows with this setup and driven the LED's without issue.

I can probably get through this year with my basic Firmware, but I really like a lot of the functions provided by the ESPixelStick firmware, so I'd love to get that working if I can.

Any suggestions you can provide or places I can look or try to tweak to diagnose or address the flickering would be greatly appreciated (I'm more than willing to dive into the code, submit PR's, etc. as appropriate).

forkineye, 2018-11-27:
So are you using a 5v buffer and the 330 ohm resistor following that or separate cases of each?  Have you tried it direct from the NodeMCU with no buffer or resistor?

redwolf3, 2018-11-27:
@forkineye 
Sorry, I missed a couple of important things:

- I have a 12v to 5v buck-converter to power the NodeMCU and high-side logic-level converter 
- I have  a 3.3v to 5v logic-level converter on the output of the NodeMCU
- The 330 ohm resistor is in-line after the logic-level converter. 

Here's what it looks like:
`NodeMCU <--> LLC <--> 330 ohm <--> ws2811 Input 1`

I tried things without the logic-level converter and resistor when writing the NeoPixelBus-based firmware and received no output if it the LLC was not present. I saw no significant difference with or without the resistor (I believe this is just here for over voltage / current protection though).

I had similar issues with flickering when  using the FastLED and NeoPixel libraries. When I was reviewing similar issues using those libraries they all reported issues with the ESP2866 modules I'm using due to the lack of hardware timers combined with the way they  bit-banged to set the timing combined with contention from the WiFi module processing. That's what initially lead me to the NeoPixelBus library which has worked flawlessly with wireless on the NodeMCU and D1 mini.

I haven't tried using a 5v buffer in-place of the logic-level converter. I can try this as well if you think it might net different results. 

NOTE: I'm on the road right now so won't be able to test anything in hardware for the next 5 days.

gjccat, 2018-11-29:
I use the follow circuit for 5v and 12v WS2811 and WS2812 Pixels, both the bullets and Strips. I run at various brightness

[image]