forkineye/ESPixelStick #144: how can I switch off the startup led?

NightD3v, 2018-12-15:
How can I turn off the startup light or that only one led lights up. Dimmed would also be a possibility.

forkineye, 2019-01-03:
This is dependent upon your board.  On an ESP-01, the red and blue LEDs are hardwired to power and serial data.

closed 2019-01-03 by forkineye

NightD3v, 2019-01-03:
Not the LED from the board... when i turn on the board all the ws2812 leds light up but i want to have these off

bquesse, 2019-01-03:
Some 'WS281x' leds that you find online are not true WS281x and include a compatible clone of the chip. This clone chip, whatever it is, has the quirk of turning all the lights on by default when they receive power but no data - usually they are blue, but this could vary. This could be what you are dealing with.