forkineye/ESPixelStick #145: WS2813B Issue with 300 LEDs (60LEDs/m, 5M roll)

GeorgeIoak, 2018-12-18:
I only have 1 of your ESPixelStick boards and it's running another strip so until you get more stock I thought I would try a strip of WS2813B that I have. They are 5V LEDs and a total of 300 LEDs

* I am using Agilent Bench Power Supply and connecting to both ends of the strip
* The same supply is powering a Wemos D1 Mini Pro v1.1
* I am running 3.1-Dev and the CPU is set at 160MHz
* The output of the D1 Mini board is fed into a SN74LVC8T245MDWEP level converter
* The output of the LLC goes through a 330 ohm series resistor which is then fed into the DI and BI pin of the strip

Using the test mode of the firmware I am always able to set the LEDs to Blue in the static mode. Usually Green works, and Red works sometimes. I have never been able to get white to work. The LEDs flicker and I'm seeing ~6A coming from the power supply. Perhaps it's a power supply issue but that doesn't explain why Blue and Green seem pretty solid but Red doesn't always work (it too will flicker). Both Blue and Green pull right about 4A but the red will flicker and pull between 4.0 and 4.5A. I assume the higher current is because it looks like some pixels are white, not red, when it flickers.

My initial testing with the strip was with the same LLC but running FastLED code on a Teensy 3.2. I was running just some quick pattern tests so I can't say for sure that a static pattern also worked with that setup.

What's most curious to me is why would Blue and Green seem to work but Red causes some trouble? Perhaps this has something to do with these being WS2813B LEDs?

forkineye, 2018-12-18:
Power issue.  Trying adjusting number of pixels to 10 and test it.  On my test matrix, I inject power every 60 LEDs.

GeorgeIoak, 2018-12-18:
I was thinking along those lines but having the trouble with the red through me off. I'll adjust down the length and let you know what I find.

GeorgeIoak, 2018-12-18:
I adjusted down to 200 and then 100 and the results are a bit mixed. I also tried 150 and depending on the number of pixels I define it determines what color flickers or gets mixed results. None of those settings ever did get white to wok though and current draw was under 5A with 150 pixels and trying to drive them white

GeorgeIoak, 2018-12-18:
I'm thinking you might be correct about the power issue. I was relying on the power supply to show me when there was too much draw but I pierced the silicon tube and measured the voltage at different points in the strip and I see voltages down to 4.0V and that's with only 200 pixels defined. Looks like I'll have to find a good method inject power through the silicon tube at multiple spots on the strip.

Not to hijack my own power but I couldn't find a description on how to use gamma and brightness. Could you point me to where it's discussed adjusting those parameters. I assume that a brightness of 1 means 100% so if I turn down the brightness I should be able to use less injection spots.

forkineye, 2019-01-03:
That is correct on brightness, however I would still do more power injection.  Enabling gamma correction adjust the dimming curves to be more natural.  I'd experiment with it and see what you prefer. I use it on all my controllers.

closed 2019-01-03 by forkineye