forkineye/ESPixelStick #146: Is there ability to add hardware WatchDog STWD100?

qawsed55, 2018-12-28:
Some comments on web says, that sometimes they have freezing light show. Not all times it is software bug.
But in any cases, I think it would be good option to have hardware WatchDog, that can reset module.
Most users use additional board to drive ESP8266. It is not problem to add on that board small WatchDog chip.
I'm not good programmer, may be some one can advise code for resetting WD chip?
Thanks. Sorry for pore English.

penfold42, 2019-01-06:
It doesn’t look too hard.
You’d need to use a gpio to enable it - the timeout would likely be breached on startup or during a firmware update

qawsed55, 2019-01-10:
STWD100PYW83F - this is hardware WDT. May be someone will make additional code for ESPixelStick
 to be able to use this WDT chip.

penfold42, 2019-01-10:
Why don’t you have a go at programming it ?

Get the one with the longest timeout (1.6s) and open drain output.
Use a spare gpio with pull-up on the enable pin
Hook it’s output to reset
Hook it’s input to the led or serial output.

The only code needed is to flip the enable gpio when the network is up

penfold42, 2019-01-10:
I think the part number is STWD100NYWY3F

closed 2021-09-24 by forkineye