forkineye/ESPixelStick #148: MQTT commands ignored

tube0013, 2019-01-06:
I'm trying to get mqtt support working within HomeAssistant.  I have it enable on the ESPixelStick:

[image]

 and can see the status messages come across when it boots up:

```
pixelstick/stick1/light/status OFF
pixelstick/stick1/brightness/status 100
pixelstick/stick1/rgb/status 255,255,255
```

in HomeAssistant I have it set up per the instructions on the main page here, only difference is the changes since mqtt-json is depreciated now:

```
light:
  - platform: mqtt
    schema: json
    name: "PixelStick"
    state_topic: "pixelstick/stick1"
    command_topic: "pixelstick/stick1/set"
    brightness: true
    rgb: true
    effect: true
    effect_list:
      - Solid
      - Blink
      - Flash
      - Rainbow
      - Chase
      - Fire flicker
      - Lightning
      - Breathe
```
When I send commands I see them come out over mqtt but the ESPixelStick does not respond to them:

```
pixelstick/stick1/set {"state": "ON"}
pixelstick/stick1/set {"state": "ON"}
pixelstick/stick1/set {"state": "ON", "brightness": 255}
```

Is there any way to de-bug this?  Any other suggestion, or pointers?

forkineye, 2019-01-06:
Wasn't aware it was deprecated.  I'll update it this week and add support for home assistant discovery as well.

tube0013, 2019-01-07:
I updated to the most recent Travis Build and mqtt is now working.  I'll keep an eye out for the update with discovery thanks!

forkineye, 2019-01-09:
MQTT Discovery for Home Assistant is working now.  Updated the README section for MQTT as well.  Let me know if you run into any issues.

closed 2019-01-09 by forkineye

towerhand, 2019-03-31:
Hello @forkineye, using the latest ESPixelStick Dev version from 12 days ago and couldn't get the discovery for Home Assistant to work. Tried with HA latest stable 0.90.2 and beta 0.91.0b3 (Hass.io and Mosquitto broker 4.1 add-on). Followed the readme but doesn't say anything about the "Home Assistant Prefix" option. Not sure if I'm doing something wrong.

towerhand, 2019-03-31:
I can confirm discovery is working on HA, it doesn't show the device information under the MQTT integrations though.