forkineye/ESPixelStick #156: Effects and Serial/DMX/Renard output.

sig-tesla, 2019-01-14:
The new Effects "tab" work great for Pixels, but for Serial/DMX/Renard it seems that this new tab appears to not have been implemented and/or wired up.   Are there plans to hook up this functionality for the Serial/DMX/Renard?

Related: I have also noticed that the baud rate selector always shows "250000" even when you try to select another option. 

Context: I would like to use the idle effect, and I would also like to control an old unused LOR DC controller since the lights in question are quite far apart pixel wise.

penfold42, 2019-01-15:
is It just the effect idle timeout that doesn’t work?
Can you manually select effects or are they broken too ?

I can already see a bug where the idle timer is ignored

sig-tesla, 2019-01-15:
It's the whole tab that seems not to work   Even selecting an effect does not light any channels/pixels.

sig-tesla, 2019-01-17:
It works great!  Thank you so much!

closed 2019-01-17 by sig-tesla

sig-tesla, 2019-01-17:
I think there is, because one might actually have RGB hooked up to those DMX channels.  We might want to default gamma & brightness to values that do nothing for DMX to avoid future questions about strange behaviour when non light items are hooked up to the DMX addresses.    e.g. "Why does my moving head jump in big increments when pointing left but smaller ones when pointing right.?"