forkineye/ESPixelStick #162: Effects: Enable on Idle doesn't work

evomotors, 2019-01-30:
Effects - Enable on Idle doesn't work. I was able to make it work only twice by switching enable/disable and some other settings on that screen, but cannot make it work again. 

This option is really important to me as I want lights to remain statically on when show is not running. Or even keep lights permanently installed year round without use of additional devices.

penfold42, 2019-01-30:
Do you have an effect selected ? A manually selected effect obviously overrides e131 data but also takes precedence over the idle effect.

What type of lights do you have ?

What version of firmware ?

evomotors, 2019-01-30:
> Do you have an effect selected ? A manually selected effect obviously overrides e131 data but also takes precedence over the idle effect.
> 
> What type of lights do you have ?
> 
> What version of firmware ?

**Effect:** Disabled
**Color:** rgb(0,0,255)
**Reverse pattern:** Off
**Mirror pattern:** Off
**All leds same:** Off
**Effect Speed:** 5
**Effect Brightness:** 1
**Enable at Startup:** Off
**Enable when Idle:** On
**Idle Timeout:** 5

**FW Version:** 3.1-dev
**Build Date:** Jan 16 2019
**Lights:** WS2811

evomotors, 2019-12-13:
Works as expected

closed 2019-12-13 by evomotors