forkineye/ESPixelStick #170: Home Assistant Integrations

towerhand, 2019-04-01:
Using ESPixelStick 3.1 Dev from March 19th and latest HA. The discovery feature works great but the device info doesn't get added to the MQTT integration and .storage in HA, limiting some of the newer functionality in HA. I think one of the reasons is because a unique_id needs to be added to each device but I'm not sure how to do that.

https://www.home-assistant.io/components/light.mqtt/#unique_id
https://www.home-assistant.io/components/light.mqtt/#device

Looks like Tasmota get it done like this and also add a sensor showing some statistics:
https://github.com/arendst/Sonoff-Tasmota/blob/7f590e62f3e5fbec230091239040e74113e70fa1/sonoff/xdrv_12_home_assistant.ino#L129

ahodges9, 2019-12-03:
@towerhand, just put up a PR to address this. Should get everything properly showing up in the integrations page.

#189

towerhand, 2019-12-03:
Awesome, thanks.

closed 2019-12-03 by towerhand