forkineye/ESPixelStick #172: Effects T6-T8 missing

sascha777, 2019-04-18:
Hi guys,
just got my ESPixelStick up and running (3.1-dev), which is really cool! Thanks for that!
I am just wondering why some effects aren't available. I saw that a fire effect should be available and can find it in the sources as well. But on the website is see only effects T0-T5. T6-T8 aren't available. Haven't tried to adress them by MQTT yet.
Any clues why they are missing on the website?

sascha777, 2019-04-18:
Tried now with v3.0 release, pre-release 20190319150609 as well with master gulped and uploaded with arduino ide.
Still can't find T6-T8 :-(

sascha777, 2019-04-18:
Tried also to find on testing page without success

penfold42, 2019-04-18:
You uploaded both the code and the spiffs parts ?

sascha777, 2019-04-18:
Yes, I did. Website looks normal, only the effect dropdown shows only 5 instead of 8 effects.

sascha777, 2019-04-18:
Found it! The effect were find until prerelease 20190116222043.
Since 20190319150422 three effects are missing.
Can't check in the code as the releases are untagged.

phazed0, 2019-06-08:
Thank you for the great work on this project!

I can confirm and have the exact problem as @sascha777. 

Further, the Chase (T5) is visible in my list but does not work (no lights on).

LarryDysart, 2019-08-31:
I found on wshandler.h:line 196, if I increase DynamicJsonDocument json(2048); it works again

forkineye, 2020-12-01:
Been fixed for a while, think back in 3.1.

closed 2020-12-01 by forkineye