forkineye/ESPixelStick #177: ESPSFlashTool dont work on my pc

maik-dmxc, 2019-06-15:
hi,

just downloaded and unziped ESPixelStick_Firmware-3.0
ESPSFlashTool.jar bring up an error

`Unable to find firmware configuration file`

`maik@maik-XMG-A505:~$ java -jar '/home/maik/Desktop/ESPixelStick_Firmware-3.0/ESPSFlashTool.jar' 
OS: linux / amd64
`

regards maik

mvinaround, 2019-08-16:
Also not working... Missing the .bin files?

tpmodding, 2019-08-20:
same problem...just startet the tool a few times suddenly it worked...

forkineye, 2019-08-20:
@maik-dmxc Run java from the same directory in which you extracted the zip file or it won't be able to find the binaries in the subdirectories. The paths referenced from within the jar file and relative and will be broken if you try to run it from another location.

leivos, 2019-09-09:
I also get "Unable to find firmware configuration file". I tried to run ESPSFlashTool.jar form the directory where ESPixelStick-master is located. I tried it also inside ESPixelStick-master directory and within dist directory.

Could some of the files be missing from the zip? I can't find spiffs/config.json, for example.

leivos, 2019-09-11:
I tried an older version (ESPixelStick_Firmware-3.0rc3). There I was able to start ESPSFalshTool.jar. and load the program but Serial Output didn't return any IP address.

Oh, and I noticed that the latest version has config.json but it's inside data/. Still can't get that new version working.

forkineye, 2019-10-22:
Unable to reproduce any issues with this.

closed 2019-10-22 by forkineye

tapi35, 2020-04-07:
Hello Leivos, do you have resolve the problem ? I'm in the same situation. I don't understand. I can flash the 3.0 version but after flashing, there's no wifi connection, ESPFlashtool don't do nothing. And when I want to use the master version, I have many errors like you. If you have resolve the problem, I'm interrested. Thanks