forkineye/ESPixelStick #182: Issue on the Travis Builds

extremeskillz, 2019-09-28:
Hi Shelby and Dev Team,
Keep in mind when building out the firmware that the latest Travis builds are not working on Linux well. What I mean is the official 3.0 build works fine whether you are using any platform with the CP2102 Module STC to upload the firmware. This works fine as the ESPixelStick used this setup. Others ESPixnodes on the market using this firmware have the integrated module so only a USB cable is needed to upload the firmware. When using Linux this isn't being seen and when it does it gives a false upload and then fails. It seems to be a driver issue but when using the CP2102 module it works since its dedicated usb hardware. Its on the built in ones like the ESP8266 NodeMCU having issue uploading firmware from Linux. Windows has tested fine and I had to resort to windows for this for ESPixNodes. When I have a chance I will see if I can get a screenshot but this is on the latest builds of this firmware not using the CP2102 to upload firmware. 

Thanks,
Carlos

forkineye, 2019-10-22:
So sometimes it doesn't populate in the drop-down and sometimes it does? And when it does, it doesn't work? What about interaction with other software, like the Arduino IDE?

extremeskillz, 2019-10-22:
Hi Shelby,
What it's doing is the node does show up in the drop to upload the image but fails to upload. Using the same travis build on Windows works fine. Seem to be and error only on Linux systems with the Tavis builds. The error I get during upload is communication error but that is all I get. Only the linux version is having this issue. Seems to be a driver issue. Using the STC module works fine since it is dedicated hardware and works as it should on your ESPixelSticks. Jboards ESPixnodes has this built in and seems to be where the issue is coming from.

extremeskillz, 2020-02-04:
To confirm: Firmware 3.1 fixes this issue with Linux uploading to the NodeMCU based pixel controllers.Based on Shelby's firmware release notes I can confirm its all working as stated in his notes on the latest pixel stick hardware based on V2 ESPixelsticks, V3 ESPixelsticks, and ESPixnodes(jBoards). Thanks for the fix.

closed 2020-02-04 by extremeskillz