forkineye/ESPixelStick #185: ws2811 flickering on wemos d1 mini

rolfberkenbosch, 2019-11-02:
Hi All,

I just saw this awesome piece of software, but my leds keeps flickering. I have set the speed of the wemos from 80Mhz to 160Mhz. But it keeps flickering. What did i miss ?

With best regards,

Rolf

penfold42, 2019-11-02:
Level shifter ?
Good wiring ?
Solid power supply - For both the LEDs and wemos ?

You haven’t detailed your hardware setup, but nearly every flicker problem can be traced to one of my questions.

forkineye, 2019-11-02:
Also, what's your network configuration? What's your e1.31 source? Try the effects section to test your hardware setup.

rolfberkenbosch, 2019-11-03:
> Level shifter ?
> Good wiring ?
> Solid power supply - For both the LEDs and wemos ?
> 
> You haven’t detailed your hardware setup, but nearly every flicker problem can be traced to one of my questions.

Do you have a good example how to make a good level shifter, I have tested my wiring and power supply and those works good. But i do not use a level shifter. I did try the effect section and that is gives flickering lights. So I think I need to use a level shifter.

penfold42, 2019-11-03:
https://hyperion-project.org/threads/level-shifter.623/

mkozik13, 2019-11-15:
Good evening,

I seem to be having this same issue.  I have tried the D1 Mini with the latest firmware.  The D1 Mini is connected directly to a matrix of 510 square pixel nodes.  I do not have a level shifter in place as I have read several others say it was not needed and if close to the first pixel, the first pixel would boost the signal moving forward.  I am powering from the same power supply and the grounds are shared.  

I also tried using wLed and this firmware does not show the same symptoms, however, it does not seem to be as responsive.  I just wanted to validate my install.  I am using xLights to drive the e1.31 signal on both.  

I would greatly appreciate any and all help with this!

Take care,

Mark

mkozik13, 2019-11-15:
I have set the speed of the wemos from 80Mhz to 160Mhz. But it keeps flickering. 

How did you change the CPU speed?  I have done a few searches but am not finding anything without going into the code.

Thanks!

Mark

forkineye, 2019-11-15:
@mkozik13 are you using the binary releases or compiling your own?

mkozik13, 2019-11-15:
Using the binary's - Have never compiled my own

forkineye, 2019-11-15:
> Using the binary's - Have never compiled my own

Then it's already set for 160MHz. Does it have the same issue if you use a rainbow from the effects tab?

mkozik13, 2019-11-16:
Yes Sir - Seems not to matter what I am doing with this one!  I put my D1 Mini with the wLed firmware and it is stable.  Switch back and flicker city!

forkineye, 2019-11-16:
> Yes Sir - Seems not to matter what I am doing with this one! I put my D1 Mini with the wLed firmware and it is stable. Switch back and flicker city!

Can you post a video of it doing the rainbow pattern? May help me zero in on what's going on.  I run it on my 720 pixel test matrix without issue so not sure what's going on.

mkozik13, 2019-11-16:
https://drive.google.com/file/d/1Hep9ZYBFY9rbJFId105SBGBkir4h3jej/view?usp=sharing

mkozik13, 2019-11-16:
It looks similar to when two instances of xLights are trying to talk at the same time really but I have made sure that no other instances of anything are 1) talking to this controller and 2) talking on this universe

forkineye, 2019-11-16:
Yes, that's very odd. Looks like two data sources but the rainbow effect would override that. What happens if you set it to a lower pixel count and only do part of the matrix? Does it still flicker if you set it for 50 pixels?

mkozik13, 2019-11-16:
Yes Sir - I have tried that as well as reducing the brightness and the flicker continues

forkineye, 2019-11-16:
Can I get a link to the pixels?

mkozik13, 2019-11-16:
https://shop.cfolights.com/product/12v-ws2811-square-modules-100ct-string-black-18awg-wire/

mkozik13, 2019-11-16:
Also, in case it helps, I am running 3.1-dev with build date of 18 April 2019

forkineye, 2019-11-16:
> Also, in case it helps, I am running 3.1-dev with build date of 18 April 2019

Trying using the one from this October - https://github.com/forkineye/ESPixelStick/releases/tag/untagged-725bf24cd612dc3ce18c.  Also, how is the Wemos being powered? By USB from your computer or by the same supply as the matrix?

mkozik13, 2019-11-16:
Same power supply as the matrix is using a 12v USB plug - Funny as I was in process of flashing the October one on another Wemos

forkineye, 2019-11-16:
Once you extract the firmware, edit firmware->firmware.json. Looks for "reset" under "devices" and change the value from "none" to "nodemcu".  Then you can flash from the flashtool without having to manually put it into programming mode. It'll work just like from the Arduino IDE.

mkozik13, 2019-11-16:
Sorry Shelby it is still there.  This is a different D1 Mini with the October firmware loaded

forkineye, 2019-11-16:
> Sorry Shelby it is still there. This is a different D1 Mini with the October firmware loaded
I'm at a loss as to why it's working with wled and no the espixelstick firmware.  I'm curious if it's a time issues, however the ws281x reset times are the same for both packages (300us).  You could try bumping WS2811_TIDLE up to 320us or 350us to PixelDriver.h to see if that changes anything. Also, have you tried by powering the wemos from your computer USB and watching the serial console? Is it saying anything?

mkozik13, 2019-11-16:
OK - This makes no sense to me but I am not the sharpest tool in the shed - When I connected the usb cable to the laptop to power the Wemos, I was getting wired results so I connected the ground pin to the power supply giving common ground.  Things were working fine, no flickering.  When I connected the usb cable back to the power block I was using (which shares the same ground) everything appeared to be working fine - Scratching my head even more, I noticed I had left the ground pin on the Wemos so I pulled it off and the flickering started again.  Put it back on and it is gone again!

So this begs the question as to why this ground pin is not needed with the wLed firmware but it IS needed for ESPixelStick, especially when wLed (from what I have read) is based off your firmware!  AND the electrical connections should not be dictated by the firmware, especially grounds - Correct?

I have been running the rainbow effect for the last 15 minutes as well as a sequence from xLights with no flickering.   

Am I wrong to say this is kinda crazy?!?!?

forkineye, 2019-11-16:
That's rather odd, but sounds like the ground potential was floating a bit when disconnected.  I just found out about the WLED project and at first glance, it looks like he's using the neopixelbus library for driving the pixels.  The uart method in that was derived from my code, but that's it. Looks like he uses my e131 library as well.  One thing I did notice though is that he seems to be running the ESP at 80MHz and I run it at 160MHz.  The extra power draw of running at a higher clock rate could possibly have an effect  on the ground potential being slightly greater and since you're already driving the pixels out of spec at 3.3v, it could've pushed it right over the edge.  Glad you figured it out though!

mkozik13, 2019-11-16:
Thanks a TON Shelby for sticking with me and helping me out with this!  Now to play with the DMX side!  Have a great weekend my Friend!

closed 2019-11-20 by forkineye