forkineye/ESPixelStick #188: Firmware upload

Mat-Moo, 2019-12-02:
Very confused ;( uploaded a bunch of firmwares and also updated a few as well all good. Now I’m setting up all my stuff I realise there are a couple of pixelsticks running old firmware and want to update them. But nothing is working now, I’ve even bought another USB adapter in case I had broken it but will not working. I’ve tried everything and fresh out of ideas. The only think I can think of now is that I updated my laptops to windows 1909 and now not compatible or something? Has anyone with latest windows run the flash java tool successfully? I’m not even getting serial data dumps or anything ;( getting pretty frustrated now and don’t know what else to do

forkineye, 2019-12-03:
Is your serial adapter showing in the dropdown from the flash tool? If not, checked device manager and windows and see if there's a driver issue.  Sure the TX and RX lines aren't swapped?

Mat-Moo, 2019-12-03:
Hiya, yep both usb/rs232 show up, the original and new one, and I've tried updating drivers on them (even though old one used to work). Yep Checked and tried both ways, but it is rx->rx and tx->tx? Hence why grasping at straws and wondering if this is a windows upgrade issue

Mat-Moo, 2019-12-07:
Would these be better for programming?
https://www.amazon.co.uk/ESP-01-Adapter-ESP8266-Wireless-4-5-5-5V/dp/B07Q17XJ36/ref=sr_1_10?keywords=USB+uart&qid=1575711142&refinements=p_76%3A419158031&rnid=419157031&rps=1&sr=8-10

Mat-Moo, 2019-12-07:
Closing, as I had ground line connected wrong, tiredness and life stress, sorry for wasting time... although the direct module upload above looks a lot less hassle!

closed 2019-12-07 by Mat-Moo