forkineye/ESPixelStick #192: Spanning universes not working

PaulBMather, 2019-12-06:
I am running build 3.1-dev on WS2811 pixels. I've got 300 pixels connected to the ESPixelStick and I'm broadcasting the first 512 channels on universe 1 and the next 387 on universe 2. Only the first universe worth of data is lighting up pixels. If I change the universe setting to 2 on the ESPixel, I can see the second universe's data lighting up my lights. So I believe what I'm broadcasting is correct (this is the same sequence I ran last year, all that has changed is the firmware on the ESPixelStick). It seems like this build is not rolling the data over to universe 2 automatically.

forkineye, 2020-01-11:
Did you set the number of pixels in the ESPixelStick configuration? It will listen to consecutive universes automatically based off your number of pixels.

forkineye, 2020-02-04:
Closed due to inactivity.

closed 2020-02-04 by forkineye