forkineye/ESPixelStick #193: Question: Can this be loaded on to MagicHome stick?

OpenLogical, 2019-12-19:
I have a couples of Magic Home ESP stick [https://github.com/arendst/Tasmota/wiki/MagicHome-LED-strip-controller], I would like to have ACN on it so I can use xlight/Vixen directly without modifying or adding hardware.  Can someone point me to the right direction?

forkineye, 2019-12-19:
No, that's an RGB strip controller.  ESPixelStick talks ws2811, GECE, DMX and Renard.

closed 2019-12-19 by forkineye

OpenLogical, 2019-12-19:
My bad on the link, it is actually a WS2811 12V controller, similar to this one [](https://www.aliexpress.com/item/32966369950.html?spm=a2g0o.productlist.0.0.10faa25cFJYMzu&s=p&algo_pvid=c68955a0-7094-4f4e-97b8-0cf1ed96f0a7&algo_expid=c68955a0-7094-4f4e-97b8-0cf1ed96f0a7-8&btsid=a7223c24-7159-4cd8-9595-aef1765eb88e&ws_ab_test=searchweb0_0,searchweb201602_8,searchweb201603_53)