forkineye/ESPixelStick #194: Failed to Program ESP8266

bretterer, 2019-12-27:
I have a NodeMCU that i am trying to flash ESPixelStick onto.  I am getting an error through the jar file.

I have (i think) put it in programming mode, by holding "flash" pressing and releasing"reset" and then releasing "flash"

I am connected to USB on my laptop.

[image]

I have tried running the commands directly in the terminal as well with no luck

forkineye, 2020-01-11:
Are you on a Mac? Make sure you have write access to the serial device.  Also, grab the v3.1 release, it will auto-reset your NodeMCU into programming mode.

bretterer, 2020-01-14:
Looks like I have the same issue with the new version.  

I am pretty sure I have write access for the serial device as well.

[image]

forkineye, 2020-01-14:
ls -l on the serial device to verify. Also, I recall another user saying you had to be a member of the wheel group I believe to write to the serial ports on mac's.

extremeskillz, 2020-01-17:
Same issue on Linux systems.

extremeskillz, 2020-01-20:
What I found was the V2 hardware required the external programing UART usb hub to connect up the hardware to configure. This was reconized by the system and you were able to configure. With the latest hardware using the ESP8266 (V3 ESPixelStick and Jboards ESPixNode) the built in usb on the ESP8266 isn't being seen on the Linux and Mac systems. This is the issue currently. Windows seems to be working fine.

forkineye, 2020-02-04:
Closed per #182

closed 2020-02-04 by forkineye