forkineye/ESPixelStick #198: Startup and Idle Effect overrides E1.31

wolf-nbm, 2020-01-20:
This could be an issue with the D1 Mini hardware I'm using, and may not effect the real ESPixelStick hardware...
If I have a Startup or Idle Effect set- it overrides everything. The device stops listening for E1.31 packets. I was hoping to use Idle Effect between shows/sequences, but once it starts- I can't send anything new to the controller. The Debug screen just shows the idle effect (I use rainbow), and the home screen's E1.31 packet counter doesn't increment with any new packets. xLights thinks everything is working fine. Testing in Multicast and Unicast mode. (I thought the latter would have a better chance of working.)
For background- I run Macs, so the precompiled Flash utility doesn't work. I compile "from scratch" in the Arduino IDE (including all of the prerequisites for the ESP8266, and gulp). The controller runs great as long as I am streaming to it from xLights- it's just the idle effects that mess with things.

penfold42, 2020-01-20:
Do you also have an effect selected from the web UI ?

Try setting it to disabled (don’t save) and wait

wolf-nbm, 2020-01-20:
I have Rainbow selected in the Web UI as an Idle Effect.
(Is there some other place to select it? Can I set one at compile time that works as-expected?)
If I set it in the Web UI and it starts- the controller (D1 Mini) ignores any E1.31 streams sent to it. The only way to override it is to go back into the Web UI and turn it off.
I can turn it off, but then I lose the Idle Effect and I would have to stream an effect between shows. I also thought it would be nice to have as a back-up in case my show computer failed for some reason.
It seems like the Startup/Idle Effect are more replacements than fillers between E1.31 streams. (I hope that makes sense.)
I'm new to all of this, so I'm likely still doing something wrong...

penfold42, 2020-01-21:
Just setting it to disabled (without saving!) should work - It then allows e131 reception and triggers the idle timer.
When the idle timer hits zero it should then use the saved idle effect

it used to but I haven’t got a recent build in front of me right now.
I will have tonight.

wolf-nbm, 2020-01-21:
I’ll give that a try. Thanks!

Wolf Butler
☁️www.nbm.com<http://www.nbm.com/>

wolf-nbm, 2020-01-21:
Okay, it's an odd fix but it does work. if I disable the effect on the Web UI but don't actually hit the Save button- it works as-expected.

penfold42, 2020-01-21:
That’s how it’s supposed to work.
You can make dynamic changes now if you manually want to set an effect.

You can then choose to save it as either a startup effect, idle effect or both

After a PixelStixk reboot and a fresh browser session, does it show “Disabled” as the selected effect ?

wolf-nbm, 2020-01-22:
I think the Save button is just confusing.
If I reboot/clear browser cache- it shows as Disabled, but still seems to be working.
Thanks again for clarifying.

penfold42, 2020-01-22:
Yep - it is a bit
The effect subsystem grew out of a “Testing” tab to let you force colours on the lights. 
As a result it’s primary use is doing things at runtime.
The save button and use for idle and/startup effects got bolted on top.

I suspect replacing the button and checkboxes with “Save as idle” and “Save as startup” would be a quick improvement.

wolf-nbm, 2020-01-22:
It is a great feature- no need to stream an idle "show" to the lights, and a nice default ("Rainbow") if there is some technical issue.

wolf-nbm, 2020-01-23:
Closing since it isn't an "issue" that necessarily needs to be resolved. Thanks for the help!

closed 2020-01-23 by wolf-nbm