forkineye/ESPixelStick #200: 3.1 parameter range and bounds

Jchuchla, 2020-01-25:
When playing with 3.1, I'm noticing that I can't tell what the bounds are on various parameters.  Especially with the effects.  It's not clear what acceptable ranges are.  It seems to allow out of bounds values without crashing, but it's not obvious when you exceed the bounds.

Also, brightness values of 0-1 are not the most intuitive.  It would be helpful (and follow expected conventions) if you could enter these as 0-100% values in the UI.

penfold42, 2020-01-25:
Here’s some:
    if (config.effect_speed < 1)
        config.effect_speed = 1;
    if (config.effect_speed > 10)
        config.effect_speed = 10;

    if (config.effect_brightness > 1.0)
        config.effect_brightness = 1.0;
    if (config.effect_brightness < 0.0)
        config.effect_brightness = 0.0;

    if (config.effect_idletimeout == 0) {
        config.effect_idletimeout = 10;
        config.effect_idleenabled = false;
    }

penfold42, 2020-01-25:
Many of the fields have a title with some guidance.
Not sure which browsers do anything with this.
Maybe a some small java cart changes to show it when clicking on the label would help

Jchuchla, 2020-01-25:
My point is that it should be shown in the UI.  Either in the field label, or a tooltip.  And all of the relative range stuff really should use the same range, ideally 0-100%.  Especially brightness.  But speed as well unless it's based on some time increment.  Let the code do the math, not the user.

MartinMueller2003, 2020-12-06:
V4 has added field validation for most direct entry parameters. There is a visible green check or a red X indicating whether the field holds a valid value.

forkineye, 2021-08-11:
We switched to percentages in V4 a while back and it's in main now. V3.2 will be the final 3.x version.

closed 2021-08-11 by forkineye