forkineye/ESPixelStick #201: Client Timeout not changing

Jchuchla, 2020-01-27:
In 3.1, i'm trying to edit the client timeout value.  It seems that no matter what I change it to, it immediately reverts to 15 when I click the save button.

I'm finding that I can't use the AP fallback feature because my AP takes longer than 15 seconds to boot and when a power failure occurs, it will always fallback to AP mode because it's not waiting long enough for the AP to come online.

penfold42, 2020-01-28:
Confirmed bug.

script.js is sending it as a string, not a number.

Can you build from source ?

Jchuchla, 2020-01-28:
I don't have the toolchain setup for it.  If you want me to test it here, you can send me an efu.  Otherwise I can wait for an update.  It's not a show stopper for me at the moment.  Just a feature I can't take advantage of.

penfold42, 2020-01-28:
Ok.

BTW, If you can set it in the config file and flash it, it should work.
It’s only a bug in the save from webUI.

commit 15463344 (2020-02-04): Fix #201 send ap_timeout as integer, not string
commit 2d2f5125 (2020-02-04): Merge pull request #202 from penfold42/fix201

Fix #201 send ap_timeout as integer, not string
closed 2020-02-04 by forkineye