forkineye/ESPixelStick #203: MQTT Birth and LWT Message

Medel, 2020-02-10:
Is it possible to add  the MQTT Birth and LWT Message to watch the state of the ESPIXELSTICK

penfold42, 2020-02-10:
Shouldn’t be too hard.

The birth message is easy - as soon as wifi and MQTT are connected, send it.

The LWT could be sent just prior to a graceful reboot.

Medel, 2020-02-10:
The LWT is send to the broker to inform the subscribers if the connection between the publisher and broker is broken. There are PINGREQ and PINGRESP packets to test the connection.

forkineye, 2020-02-11:
I'll have to look into this further. My concern would be the broker disrupting sticks / injecting wifi lag during a show.  Recommended show implementation is a dedicated multicast environment to minimize any handshaking that may cause packets to back up in the access point.  One outlying stick that requires a few retries to successfully respond to a MQTT ping could affect the entire show.

Medel, 2020-02-11:
The stick has to send the PINGREQ and the broker has to respond. The stick only sends once in the keep alive period.

forkineye, 2020-02-11:
That shouldn't too big of a problem then. We would just need to be smart about (not pinging during a running show) or warn the users of potential impacts if MQTT is enabled.  I'll flag it as an enhancement for 4.0 which is currently being worked on in the unify branch.

travisghansen, 2020-06-21:
LWT is completely handled by the broker no? I think and req/resp is likely already going on behind the scenes anyhow so I don’t think this would impact anything from a performance perspective.

It would be nice to independently set the topics for this from the rest of the mqtt topics as I control a bunch of the devices via a single mqtt topic.

MartinMueller2003, 2020-11-07:
@forkineye, I dont see anything called a birth message. There is an explicit connection made to the broker at start. is there anything more that needs to be done?

Medel, 2020-11-07:
I don't know exactly how it works, but I found a page that has a little description.
https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

MartinMueller2003, 2020-11-07:
Not sure what problem this solves. It does not look difficult to do.
Martin

 
 

https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

travisghansen, 2020-11-07:
In my case it’s a bit ugly. I want to detect when the device goes offline and reset a wall plug to cycle the power on the device to make it reconnect. I know it’s supposed to stay connected buuuut.

More generally I can see several use cases for knowing if the device is online or not.

MartinMueller2003, 2020-11-07:
So you are going to generate a LWT message that gets sent to the plugs to tell them to cycle power. Is there not a way to do that using a script that fires when the broker detects the device has gone off line (I do not use MQTT so I do not know what it can do)?
Martin

travisghansen, 2020-11-07:
LWT is the mechanism brokers use to notify of devices offline. In my case I’d create a flow in Node-RED that ‘listens’ to the LWT and subsequently sends another command over mqtt to my smart plug to power cycle. The device would reboot and be happy and functional again.

It’s ugly, but would be effective against reality. Ideally my devices would reconnect nicely and not freeze..but they disconnect and freeze over time. It’s a short term solution in this situation.

Medel, 2020-11-07:
see: []https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md
`AsyncMqttClient& setWill(const char* topic, uint8_t qos, bool retain, const char* payload = nullptr, size_t length = 0)`

example:
```
const char mqtt_will_topic[] = "diy/esps/3e7f35/LWT";
const char mqtt_will_payload[] = "OFFLINE";

```
in:  // Setup MQTT Handlers
`mqtt.setWill(mqtt_will_topic, 2, true, mqtt_will_payload, 0);`

The brocker sends "OFFLINE" when the device not sends something
if it connects it should send "ONLINE" to set the state

travisghansen, 2020-11-08:
Is there something in code that explicitly sends the online message? That’s not implicitly a feature of lwt in the spec.

The other nuanced thing to consider is that (based on my observations and understanding) lwt is only sent by the broker for unclean disconnects. So a device that cleanly goes offline won’t result in an lwt message being sent. Some apps send the same message on disconnect themselves to provide a consistent behavior in both scenarios...but that’s not part of the spec/protocol and would be implemented in the firmware itself.

MartinMueller2003, 2020-12-01:
The MQTT library used by the Unify branch has an API for setting the LWT. I added an LWT string field on the MQTT UI page and added the code needed to set up the LWT message. I have no idea what the library does with this information. You can find the code on the "https://github.com/MartinMueller2003/ESPixelStick/tree/unify" branch. It would be great if someone that understands/uses MQTT (that would not be me) could test this.

Medel, 2020-12-13:
I tried to compile https://github.com/MartinMueller2003/ESPixelStick/tree/unify but there was an error secret.h not found.
After I create the file with the SECRETS_SSID and SECRETS_PASS it compiles
I upload the program and the data. The device gets the IP from the AP but there is no page on it. -> Page Not found

Output from the device:

> WiFi Entering State: Connected To AP
WiFI: ERROR: STATIC SELECTED WITHOUT IP. Using DHCP assigned address
WiFi Connected with IP: 192.168.1.51

forkineye, 2020-12-13:
You have to compile with web stuff with gulp and upload with LittleFS. All in the readme.

Medel, 2020-12-13:
I did exactly that.
This is the output from gulp:

> [14:28:50] Using gulpfile /windows/D/Eigene Dateien/Ardurino/ESPixelStick/gulpfile.js
[14:28:50] Starting 'default'...
[14:28:50] Starting 'clean'...
[14:28:50] Finished 'clean' after 26 ms
[14:28:50] Starting 'html'...
[14:28:50] Finished 'html' after 351 ms
[14:28:50] Starting 'css'...
[14:28:51] Finished 'css' after 562 ms
[14:28:51] Starting 'js'...
[14:28:55] Finished 'js' after 4.45 s
[14:28:55] Starting 'image'...
[14:28:55] Finished 'image' after 21 ms
[14:28:55] Finished 'default' after 5.42 s

and this the output from the Arduino IDE:

> Executable segment sizes:
IROM   : 466280          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28600   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1448  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 7040  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 32312 )         - zeroed variables      (global, static) in RAM/HEAP 
Der Sketch verwendet 503368 Bytes (48%) des Programmspeicherplatzes. Das Maximum sind 1044464 Bytes.
Globale Variablen verwenden 40800 Bytes (49%) des dynamischen Speichers, 41120 Bytes für lokale Variablen verbleiben. Das Maximum sind 81920 Bytes.
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 80:7d:3a:3e:7f:35
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 507520 bytes to 352852...
Wrote 507520 bytes (352852 compressed) at 0x00000000 in 8.1 seconds (effective 502.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[SPIFFS] data    : /windows/D/Eigene Dateien/Ardurino/ESPixelStick/data
[SPIFFS] size    : 2024
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/config.json
/www/alexa.html.gz
/www/ddp.html.gz
/www/disabled.html.gz
/www/dmx.html.gz
/www/e1_31.html.gz
/www/effects.html.gz
/www/esps.css.gz
/www/esps.js.gz
/www/fpp_remote.html.gz
/www/gece.html.gz
/www/index.html.gz
/www/mqtt.html.gz
/www/relay.html.gz
/www/renard.html.gz
/www/serial.html.gz
/www/ws2811.html.gz
[SPIFFS] upload  : /tmp/arduino_build_228598/ESPixelStick.spiffs.bin
[SPIFFS] address  : 0x200000
[SPIFFS] reset    : --before default_reset --after hard_reset
[SPIFFS] port     : /dev/ttyUSB0
[SPIFFS] speed    : 921600
[SPIFFS] python   : /home/mario/.arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3
[SPIFFS] uploader : /home/mario/.arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/upload.py

esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 80:7d:3a:3e:7f:35
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 2072576 bytes to 86686...
Wrote 2072576 bytes (86686 compressed) at 0x00200000 in 2.1 seconds (effective 7805.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

forkineye, 2020-12-13:
You're still using SPIFFS. Go through the readme again. Me moved a LittleFS a while back.

Medel, 2020-12-13:
Thank you. I change to  Littlefs and it works

closed 2020-12-21 by forkineye