forkineye/ESPixelStick #204: Setting up multiple universes

radioredman, 2020-03-19:
I am wondering how to set up multiple universes... Do I set the number of universes with the web interface under "Universe"?
[image]

forkineye, 2020-03-19:
It will listen for consecutive universes starting at "Universe" based upon your universe boundary.  In your screen shot for example, you have 600 pixels (1800 channels).  The first 510 channels (170 pixels) would be on Universe 4, the next 510 on 5, then 6, and the rest on 7.

radioredman, 2020-03-19:
Thank you! I appreciate your response and the amazing work on this! This
clears up my confusion!

closed 2020-03-19 by forkineye