forkineye/ESPixelStick #207: Start Channel Config

tinseltunes, 2020-04-09:
I not as issue as such, just some advice.
Should the Start channel always start at 1 or the actual start channel i.e. 225 if the model start channel is 225 in this case.

And should the universe boundary be 510 or the actual number of channels based on the pixel count i.e 64 pixels = 192 channels = Universe boundary.

Thanks
in advance.

mkozik13, 2020-09-09:
Cannot comment on the DDP as I still have not traveled down that path, however, mine are all set to the correct start channel shown in xLights along with the universe and I set my boundries to 510.  Hope it helps!

forkineye, 2020-10-27:
Sorry I missed this.  The start channel is the offset into the universe you want to start listening at for that controller, and the universe boundary should match whatever you're using in your sequencer. 512 is recommended and you want to pack each universe full of data.

closed 2020-10-27 by forkineye