forkineye/ESPixelStick #212: Parallel output , direct port access

Kamal-Sonani, 2020-09-01:
Instead of using one single line of output , is their any changes  to take 4 parallel output (using direct port access).it Will take same time as single pin output.

On each port 500 led (512 ideally) x 4 port = 2048 led

By doing that, frame rate will be increased instead of driving  >600 led on single line.

MartinMueller2003, 2020-11-07:
I have been looking into this and it can be done since the I2S interface can support better than 10Mb transfer rates (we need 3Mb/output port). This would also require additional external HW (beyond a level shifter). It is up to Shelby to cecide if we want to support this feature.

forkineye, 2020-12-01:
It's currently outside the scope of ESPixelStick as it would require specific external hardware to be supported that's apart from the ESPixelStick hardware.

closed 2020-12-01 by forkineye