forkineye/ESPixelStick #213: Question : Trigger Relay while in E131 mode

mkozik13, 2020-09-05:
Have a need for Halloween to install an ESPixelStick in a prop using E131 driving pixels and require a relay within that I can control via xLights.  Can this be done today?

MartinMueller2003, 2020-11-07:
Sorry, I did not see this sooner. Yes we could add a relay on/off output module. The relay control would be "Just another byte" in the existing E1.31 data stream.

While this can be done, it is up to Shelby to decide if we should add this feature

mkozik13, 2020-11-07:
No worries Martin - Thanks for the reply.  Would be awesome if possible!

forkineye, 2020-11-07:
This is something I was wanting to implement this season as well as I have a few props I'd like to trigger in my Halloween show. Just didn't have time to do it. For configuration, I'd have a threshold value for what determines on/off and output pin configuration. May need to be a safety timer that only allows relays to trigger so quickly so they don't get trashed by someone running a quick pulsing effect on them by accident.

MartinMueller2003, 2020-11-07:
This should not be difficult. I will start to add it. It will support a fixed number of output pins. 

Feature List:
configurable output pin
max refresh rate
trigger level for "on/off"
Invert / non invert (aka on == output low /  on == output high)

Anything I missed?

mkozik13, 2020-11-07:
Not trying to be greedy, just asking - Is the intent to allow for a single relay or multiple?  Knowing the Arduino relay modules are single, doubles, 4 bangers and 8's?

MartinMueller2003, 2020-11-07:
A number of factors control this. I will get as many as I can on the LoLin Pro ESP8266 and ESP32 platforms (those are what I have) :)

mkozik13, 2020-11-07:
I would think the dual relay module would be fine for what folks do - If not, they can always add another D1!!

MartinMueller2003, 2020-11-13:
On my branch I have implemented an 8 channel version. Please take a look at the UI to see if this is what you want to see. I have not tested the functionality yet (though it is all implemented). I am seeing a random crash on the ESP8266. Have not chased it down yet.

MartinMueller2003, 2020-11-22:
This code is now available on the Forkineye unify branch. Can someone other than me test it and then we can close this issue.

patdelaney, 2020-11-22:
I may be able to take a look, I have some single relay shields (if that is the correct term )

patdelaney, 2020-11-22:
is it in the latest travis build?

patdelaney, 2020-11-22:
I have a D1 module and a relay shield. I know from a previous project the
relay used the D1 pin.  How do I configure it in the Unify UI?

MartinMueller2003, 2020-11-22:
On the UI go to the device configuration. Select relay from the drop down. This brings up the relay config. For channel 1, set a GPIO pin you want to use and enable it. Any data sent to the ESP should trigger the relay output.
Martin

MartinMueller2003, 2020-11-28:
After a meeting between Pat and I, we got him set up and working. At this point we can consider this feature request completed.

forkineye, 2020-12-01:
Closing per @MartinMueller2003

closed 2020-12-01 by forkineye