forkineye/ESPixelStick #221: AP Fallback not working

bwerst, 2020-10-26:
Been enjoying ESPixelStick for several years and liking very much.  Last week bought two of your version 3 hardware from Amazon.  They work great in my home on my Wifi which is in the config.json file.

However, I'm trying to use them at my daughters school for upcoming school play.  The AP fallback isn't working.  My phone will connect to the ESPixelStick.  It shows my phone IP is 192.168.4.2 with a gateway of 192.168.4.1.  Switch to Chrome to access the web interface and nothing.  http://192.168.4.1 doesn't work, http://null which for some reason defaulted as hostname doesn't work.

I did initially boot on my wifi and then went in checked the AP failback option under Wireless settings.

Also, I'm not sure where the .4.x subnet is coming from.

forkineye, 2020-10-27:
What phone are you using, and are you on 3.1? On my Android (Moto G7), I get a notification that there isn't internet access and I have to tell it I want to stay connected.  Also, make sure you're connecting via http and not https.

bwerst, 2020-10-27:
Hi, 
Thank you for the follow up.  I'm on Android OnePlus 5.  The phone seems to stay connected.  Tried both http & https.

Where would I set what IP address I want the ESPS to use in AP fallback mode?  Slight chance I'm using a hybrid 3.1.  I copied a previous project that might have been 3.0.  I manually overwrote each of the files that had newer revision dates on github.  Your update .jar tool wouldn't find files.  

I used Arduino IDE to compile and upload firmware.  Then Upload sketch feature to get www files loaded.  Unit works great on my WiFi at home, just not in fallback in the wild.  Yesterday I took my actual home WiFi AP into the school and plugged into the school LAN for DHCP.  Used a network scan app on my phone to find the IP address of the D1 mini.  I would have saved time if I'd remembered to record the mac address ahead of time.

I was thinking about entering a list of SSIDs including the schools so it will hopefully come up.  Haven't tried it yet.  Also need to see if I can get a right angle usb adapter.  The capacitor and fuse on your board block the usb port when assembled.  Was hoping to use serial monitor to confirm settings.  Do you have a version three schematic published?  I'm curious to use dual sided pins on a wemos in order to connect a TFT display panel to show IP and universe information. 

Thanks!

forkineye, 2020-10-27:
You can see the version information under the Admin tab of the web interface as well.  The only issue I'm aware of with the flash tool is that it has issues under newer versions of MacOS.  Are you extracting the zip archive before you attempt to run ESPSFlashTool.jar?  As for the 192.168.4.1 address, it's just the default provided by the ESP8266 Arduino Core.  I don't provide a configuration point for it, but you could modify the code if needed. Info on it here - https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/soft-access-point-class.html.  I don't think I have V3 schematic posted anywhere yet. As for Wemos shield stacking, it works great if you use a module with the stacking headers. If you have extra shield headers, you can use those to give the socket some height and get access to USB for developing as well.

bwerst, 2020-11-05:
Thanks for the stacking headers suggestion, such an obvious thing you don't always think about.  As for the default AP mode, I figured out a completely different solution for the school play.  Thank you for your time and I look forward to using units I purchased at home.

closed 2020-11-05 by bwerst