forkineye/ESPixelStick #224: WiFi Connection issues

mkozik13, 2020-11-04:
Running 3.1 and have experienced several power outages the past few days with the hurricane and all and each time I had to manually reset the D1 mini to get it to connect to the WiFi.  The fallback AP was there, I could connect to that and then perform a reset, but was wondering if it were possible to maybe extend the time or number of times the unit tries before falling back?  I am guessing the D1 Mini's are coming online way before the Unify AP is and timing out too soon.

Thanks in advance,

Mark

forkineye, 2020-11-06:
There's a timeout value for AP association named "sta_timeout" under the network section in config.json.  If the sta_timeout key isn't there, it defaults to 5 seconds.  You can increase it to see if it helps, but I would recommend not using a compiled / fallback AP.  This can be especially bad during a show as if a module loses association for some reason, it would re-associate with the wrong AP.

mkozik13, 2020-11-06:
Sorry Shelby, guess I am not smart enough to understand ‘not using a compiled / fallback AP?  I always have that checked in the UI as many of my D1 minis are inside enclosures and the reset button is not readily available.  I run all of these on their own network with a single Ubiquity AP.  Not good?

forkineye, 2020-11-06:
Oh, sorry. I thought you were talking falling back to a compiled AP.  You're talking about the "AP Fallback" option configured in the UI.  I recommend you leave that unchecked.  It's really only useful for when you know you're going to be switching APs. For instance, you're changing to a different AP or possibly taking the device somewhere where the AP credentials are unknown.  For show / permanent usage, you want to leave it unchecked. ESPixelStick will continually try to re-associate to the configured AP endlessly, rebooting as needed.

mkozik13, 2020-11-06:
OK - So right now the option is checked, is that the reason why it is not retrying to connect and requiring a reset?  I see the AP and can connect and reset it that way.  If I uncheck the "AP Fallback" then if the device comes online before the AP does, it will continue to try to connect and net require me manually resetting?

forkineye, 2020-11-06:
> OK - So right now the option is checked, is that the reason why it is not retrying to connect and requiring a reset? I see the AP and can connect and reset it that way. If I uncheck the "AP [...]

That's correct. What's happening now is the modules are timing out trying to associate with your AP so they start up as their own AP. If you uncheck the option, they'll just continue resetting and attempting to connect to the configured AP.

mkozik13, 2020-11-06:
Sweet - I will make the change.  Thanks Shelby!  Happy weekend!

forkineye, 2020-11-06:
> Sweet - I will make the change. Thanks Shelby! Happy weekend!

Thanks, you as well!

closed 2020-11-06 by forkineye