forkineye/ESPixelStick #225: Unify Branch: Network Settings not saving.

neowolfwitch, 2020-11-05:
I was able to compile the Unify Branch after installing the LittleFS ESP32 library, updating the ESP32fs plugin to the "lorol" variant and adding the mklittlefs to the tools. I also updated my SSID and password in WiFiMgr.cpp.

Everything appears to be working correctly except for the Network Settings, which cannot be changed. I can't set a different SSID/Password, Hostname, or a fixed IP address. The Timeout settings can't be set, nor can the AP Fallback box be checked and saved. 

Serial monitor shows: Configuration file: '/config.json' saved.
Then the MCU reboots back to the original settings.

forkineye, 2020-11-05:
A few things are currently broken in Unify as we're transitioning some things. You may want to drop a commit or two back for something that works right now. Hoping to have it fixed today

neowolfwitch, 2020-11-05:
Thanks! Just wanted to give you some feedback. I know it’s Alpha right now.

forkineye, 2020-11-20:
Haven't tested on ESP32 yet, but it's working on ESP8266. Also, Unify is now triggering Travis-CI builds for ESP8266. Binaries are available under releases.

closed 2020-11-20 by forkineye

neowolfwitch, 2020-11-20:
Thanks for the update! I will check it out.