forkineye/ESPixelStick #226: esp avec rasberry et fpp falcomplay

lassauce, 2020-11-15:
la fonctionne proxy pour les esp ne fonctionne pas je m'explique quand je suis sur mon Raspberry pi avec le ffp jai une option proxy qui me permet de rentre dans mes esp ou autre sans changer de réseaux mai quand je suis sur mon Raspberry et je mai adresse ip de esp en mode proxy jai aucune connections pouvez vous voir pk merci

forkineye, 2020-11-20:
J'utilise Google Traduction. Quelqu'un peut-il mieux traduire cela pour moi?
I'm using Google Translate. Can anyone translate this better for me?

the proxy works for the esp does not work I explain myself when I am on my Raspberry pi with the ffp I have a proxy option which allows me to enter my esp or other without changing networks May when I am on my Raspberry and I may ip address of esp in proxy mode i have no connections can see you pk thanks

patdelaney, 2020-11-21:
I think i understand the report, but before I try to explain it, I'll see
if I can reproduce it

lassauce, 2020-11-22:
je ne peut pas aller dans les esp par le proxy de fpp

lassauce, 2020-11-26:
WebSocket connection lost, attempting to reconnect... voila ce qui s'affiche quand je veut fair un proxy

XBouBou, 2020-11-28:
He doesn't reach correctly ESPixelStick through the Proxy of FPP (Falcon Player ).

On the browser a message "WebSocket connection lost, attempting to reconnect..." appear from the ESPixelStick

MartinMueller2003, 2020-11-28:
This sounds like an issue in setting up the IP address domain in the FPP.

patdelaney, 2020-11-29:
Martin, I think I know what he is reporting, I just have not had a chance
to reproduce.

patdelaney, 2020-11-29:
so that I can explain the scenario

lassauce, 2020-11-29:
moi jai le souci avec tt mai esp

closed 2021-06-29 by lassauce