forkineye/ESPixelStick #231: Upload tool for 3.2 not working for Ubuntu 20.04lts

extremeskillz, 2020-11-22:
Hi Shelby,
The 3.2 uploader is not working with JBoards ESPixnode for some reason. It says it can't see the serial device even though it does show up in the java flash tool. I reverted back to the 3.1 uploader and that works fine. 

Ubuntu 20.04LTS
Attempted to use the 3.2 upload tool but reverted back to 3.1. 
Attached is the screenshot of the 3.2 upload tool error.

[image]
3.2 firmware itself is fine after uploading it to the ESPixnode post 3.1 install.

Carlos

forkineye, 2020-11-22:
@extremeskillz is it possible to get a paste of the errors in the terminal window behind the GUI?  Looks like it can't find python for some reason. You have Python 3 installed?

forkineye, 2020-12-01:
Closing due to no response.

closed 2020-12-01 by forkineye

extremeskillz, 2020-12-05:
Hi Shelby,
Sorry for the late response. I had to rebuild one so I figured I paste that request to you. Here is the error in the CLI you requested using the 3.2 uploader.
carlosjr@hplaptop1:~/Downloads/Linux Software/ESPixelStick_Firmware-3.2$ sudo java -jar ESPSFlashTool.jar 
[sudo] password for carlosjr: 
OS: linux / amd64
java.io.IOException: Cannot run program "python": error=2, No such file or directory
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.exec(ESPSFlashToolUI.java:470)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.doInBackground(ESPSFlashToolUI.java:522)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.doInBackground(ESPSFlashToolUI.java:450)
	at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
	... 10 more
java.io.IOException: Cannot run program "python": error=2, No such file or directory
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1128)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1071)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.exec(ESPSFlashToolUI.java:470)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.doInBackground(ESPSFlashToolUI.java:528)
	at com.forkineye.espsflashtool.ESPSFlashToolUI$ImageTask.doInBackground(ESPSFlashToolUI.java:450)
	at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.base/java.lang.ProcessImpl.forkAndExec(Native Method)
	at java.base/java.lang.ProcessImpl.<init>(ProcessImpl.java:340)
	at java.base/java.lang.ProcessImpl.start(ProcessImpl.java:271)
	at java.base/java.lang.ProcessBuilder.start(ProcessBuilder.java:1107)
	... 10 more

Again, I run the 3.1 tool using the same steps and no errors. 

Carlos