forkineye/ESPixelStick #233: Invert setting doesn't appear to work for relay configuration.

patdelaney, 2020-11-24:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
Travis Ci Build ESPixelStick_TRAVIS-20201124014949

**Hardware Version**
Memo D1 Mini with Relay hat  Triggering on Pin 5

**Binary release or compiled yourself?**
Travis Build

**Operating System**

**Access Point**

**Describe the bug**
I configured the relay as follows:
Primary input: E1.31              Secondary input : Disabled
Primary Output: Disabled       Secondary Output: Relay

Relay Config.
Ch 1   Enabled        Inverted (NOT CHECKED)     GPIO ID: 5     Trigger: 200

Ch 2-8 Not configured

Click save Relay engaged
Disable Secondary Output:  Click save  Relay turns off.
Re-enable relay
Use Da_tester tool got to fader and slide ch1 up
When out reached 200 relay shuts off / drop below 200 relay engages.

If I toggle invert on it has no effect on the state of the relay.

[image]

MartinMueller2003, 2020-11-28:
Working on this. Found an issue in the Web UI code. New version soon.

MartinMueller2003, 2020-11-28:
New code has been pushed and submitted to the unify branch. There was a data corruption in the UI that caused bad information to be sent to the ESP. There was an issue in the ESP that processed the bad data incorrectly and really messed up the stored config. Any existing config needs to be completely replaced.

patdelaney, 2020-11-29:
FYI I appreciate the explanation, rather than.. "fixed issue".   As a
tester know what was wrong and how you fixed it makes testing easier so we
know what changed and what potential side effects that may or may not
cause.  Thanks

MartinMueller2003, 2020-11-29:
As a developer for 45 years, good explanations got drilled in early :). I have also managed QA teams and bad data always caused me to have a discussion with developers.

MartinMueller2003, 2020-11-30:
Pat messaged me indicating the invert output option is working on the MartinMueller2003 branch. The pull has been incorporated into the forkineye branch.

forkineye, 2020-12-01:
Safe to close this then?

MartinMueller2003, 2020-12-01:
Yes

closed 2020-12-01 by forkineye

patdelaney, 2020-12-01:
The fix for invert works :-)