forkineye/ESPixelStick #234: Travisbuild on Wemo D1 not stable  it spontaneously reboots

patdelaney, 2020-11-24:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
ESPixelStick_TRAVIS-20201124014949.zip

**Hardware Version**
D1 mini clone
**Binary release or compiled yourself?**
Binary release
**Operating System**

**Access Point**

**Describe the bug**
The pixels Di mini with a single relay hat spontaneously reboots, the use page is usually open, and I also have the espflashtool connected so I can see the serial output.  Is the same behavior that Martin witnessed on the zoom call Sunday.  Since I'm new to this environment I'm open to suggestion on how to gather more data for testing.

It appears that the unstableness is related to the Device setup page

mixchief, 2020-11-24:
ESPixelStick Firmware Version
ESPixelStick_TRAVIS-20201124014949.zip

Hardware Version
NodeMcu Lua V3 ESP8266 CH340G ESP-12E 
Binary release or compiled yourself?
Self compiled 

Describe the bug

As Pat describes, reboots after making changes.  Using the serial monitor in the arduino IDE it looks like the config json files are not writing correctly.  If I give the device a static ip (removing DCHP) the device fails to connect and tries to use default ssid etc.  If I upload the gulped files via the littlefs plug in the device reconnects ok.  I did manage to make changes to the device setup  but was inconsistent in saving some of the settings.  I wasn't sure if I had a device with dodgy flash, was going to try a different device this evening.

I know Pat has covered this in another post, inverting the relay outputs does not seem to function.

I too am new to this environment I'm open to suggestion on how to gather more data for testing.

MartinMueller2003, 2020-11-25:
I also see an occasional reboot right after I save a config. This is usually a ram issue. I am not seeing random reboots while data is being received. This is tested on my LoLin D1 and E32 Pro.

MartinMueller2003, 2020-11-27:
Just to check, I just let my D1 Mini run for three uninterrupted days. No crashes, no issues. I can still connect to them via the web UI. 

FYI: The Travis build is NOT designed for the D1 Mini. The code gets loaded into the wrong location. You need to use the LoLi D1 & Mini board setting.

MartinMueller2003, 2020-11-30:
With lots of monitoring I noticed that I had a hang and WDT reset after issuing the get options request from the UI. Removed the options reporting mechanism and replaced it with a scan of the configuration data that the UI already has. Have not seen the reset after config change since this change was made. The get options code was a RAM intensive operation that would get stuck if there was not enough ram available to generate the options report.

patdelaney, 2020-11-30:
Cool,, I was busy today at work so I have not had time to pull the logs
from my test and toss them into excel. I'll try tonight or tomorrow,  I am
currently working an issue with the fpp developers

Pat

MartinMueller2003, 2021-03-04:
Not seeing this anymore. I suggest we close this.

closed 2021-09-23 by forkineye