forkineye/ESPixelStick #235: Add Save As button to bottom of ESPFlashtool to the serial output section section

patdelaney, 2020-11-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Add the ability to save the serial output to a file file

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Add the ability to save the serial output to a file file so that we can send the logs to developers

**Additional context**
Add any other context or screenshots about the feature request here.

MartinMueller2003, 2022-03-16:
Added a button that saves both the System and Serial logs at the same time and in the same file.

MartinMueller2003, 2022-03-21:
This is working in the latest version. A user has generated a log and attached it to another issue.

closed 2022-03-21 by MartinMueller2003