forkineye/ESPixelStick #241: Add build number to FW Version in the Admin section of the Webui and also supply that back in FPP discovery

patdelaney, 2020-11-26:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I have several types of pixel sticks that I am testing Unify branch on. Some of my testing is in conjunction with FPP integration.  When I look at the Multisync page in FPP is only displays 4.0_unified-dev as the version, so does the ESP web ui. That information is not really useful. Is there a way that we can add a build number or the Travis number to that field. It would be a lot more useful. I would also light to know how to set that value if I compile the source myself, Ideally I want to be able to Identify what version of the branch I have. The developers do something similar in FPP. (I don't know how they do it though.)

[image]

[image]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Additional context**
Add any other context or screenshots about the feature request here.

forkineye, 2020-12-01:
The FW version doesn't increment for the auto-generated Travis builds. We leave it like that so that there's no confusion between a stable release and a Travis build.  The build date can be referenced, but you won't see it in FPP.  We update the version manually once we reach a stable release.  For 4.0, you can change it in the main sketch - https://github.com/forkineye/ESPixelStick/blob/unify/ESPixelStick.ino#L84. For 3.x, it's in ESPixelStick.h.  I'll take another look at this once we migrate to GitHub Actions for CI and see if we can include append the commit hash or something to uniquely identify it.

patdelaney, 2020-12-01:
How about this for the interim:   const String BUILD_DATE =
String(__DATE__) + " - " + String(__TIME__);

[image: image.png]

forkineye, 2020-12-01:
> How about this for the interim: const String BUILD_DATE = String(__DATE__) + " - " + String(__TIME__); [image: image.png]
> […](#)

We can do that.

patdelaney, 2020-12-01:
I tested it locally and it seems to be working. The only other thing if you
can pass something in at build time to indicate it's a travis (automated)
build rather than local development build. Meaning I just pulled the
latest source and compiled it myself.

MartinMueller2003, 2020-12-15:
Added code to report on FPP:

[image]

closed 2020-12-21 by forkineye