forkineye/ESPixelStick #244: V3.2 Flash Tool fails under Win10

packetbob, 2020-12-01:
**ESPixelStick Firmware 3.2**

**ESPixelPOPs with ESP01 module**

**Binary**

**Win10**

**Cisco E1200**

**V3.2 Flash Tool fails to load firmware**

I'm having an issue with the 3.2 Flash Tool on Win10..
The new 3.2 Flash Tool fails...
The previous 3.1 Flash Tool works fine...
Both were done running from the extracted folders..
I have the Status info from both below:

################################################## ################################################## #########
# Using 3.2 Flash Tool & Firmware:
################################################## ################################################## #########

-= Building SPIFFS Image =-
bin/win32/esp8266/mkspiffs -c spiffs/ -p 256 -b 4096 -s 131072 firmware/spiffs.bin
/config.json
/www/esps.css.gz
/www/esps.js.gz
/www/index.html.gz
Done

-= Programming ESP8266 =-
bin/win32/python3/python bin/upload.py --chip auto --baud 115200 --port COM4 erase_region 0xDB000 131072
esptool.py v2.8
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:d1:9e:d2
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 1.0 seconds.
Hard resetting via RTS pin...
Done
bin/win32/python3/python bin/upload.py --chip auto --baud 115200 --port COM4 --before default_reset --after hard_reset write_flash 0x0 firmware/pixel-3.2-1m_128k.bin 0xDB000 firmware/spiffs.bin
esptool.py v2.8
Serial port COM4
Connecting........_____....._____....._____.....__ ___....._____....._____....._____
Traceback (most recent call last):
File "C:\Users\Bob\Desktop\PixelWorkshop\ESPixelStick_F irmware-3.2\ESPixelStick_Firmware-3.2\bin\upload.py", line 19, in <module>
esptool.main(sys.argv)
File "C:/Users/Bob/Desktop/PixelWorkshop/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 2883, in main
esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
File "C:/Users/Bob/Desktop/PixelWorkshop/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 274, in detect_chip
detect_port.connect(connect_mode)
File "C:/Users/Bob/Desktop/PixelWorkshop/ESPixelStick_Firmware-3.2/ESPixelStick_Firmware-3.2/bin/esptool\esptool.py", line 483, in connect
raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to Espressif device: Timed out waiting for packet header
Done

*** PROGRAMMING FAILED ***

################################################## ################################################## #########
# Using 3.1 Flash Tool & Firmware:
################################################## ################################################## #########

-= Building SPIFFS Image =-
bin/win/mkspiffs.exe -c spiffs/ -p 256 -b 4096 -s 131072 firmware/spiffs.bin
/config.json
/www/esps.css.gz
/www/esps.js.gz
/www/index.html.gz

-= Programming ESP8266 =-
bin/win/esptool.exe -cd nodemcu -cb 115200 -cp COM4 -ca 0x000000 -cf firmware/pixel-3.1-1m_128k.bin -ca 0xDB000 -cf firmware/spiffs.bin
Uploading 445040 bytes from firmware/pixel-3.1-1m_128k.bin to flash at 0x00000000
.................................................. .............................. [ 18% ]
.................................................. .............................. [ 36% ]
.................................................. .............................. [ 55% ]
.................................................. .............................. [ 73% ]
.................................................. .............................. [ 91% ]
................................... [ 100% ]
Uploading 131072 bytes from firmware/spiffs.bin to flash at 0x000DB000
.................................................. .............................. [ 62% ]
................................................ [ 100% ]

-= Programming Complete =-

forkineye, 2020-12-01:
I see the issue now. In 3.2, we erase the chip before flashing it. This requires putting the chip into flash mode a 2nd time for programming.  This isn't an issue on the ESPixelStick V3 hardware since the Wemos module has auto-reset circuitry, however it will cause issues on older ESPixelStick V1/V2 boards and other board which utilizes ESP-01 modules. The fix for now would be to flash 3.1 and load the 3.2 EFU, however 3.2 doesn't provide any fixes or new features. It was mainly to update the flash tool to support MacOS users on Catalina.

packetbob, 2020-12-01:
Thanks for the update.  Is it safe to assume there will be no further development on code for the ESP-01 hardware so any existing users should stick with the 3.1 code? If that's the case then I will get the ESPixelPOPs info page updated to reflect this..Thanks,Bob

forkineye, 2020-12-01:
> Thanks for the update. Is it safe to assume there will be no further development on code for the ESP-01 hardware so any existing users should stick with the 3.1 code? If that's the case then I will get the ESPixelPOPs info page updated to reflect this..
Thanks,Bob

Well, it needs to be fixed as 3.x is meant to load on an ESP-01.  Moving forward with 4.x however (under development in the Unify branch), the ESP-01 will not be supported as we've outgrown the flash available.  For now, ESP-01 users should stay on 3.1 until I can put out a 3.x release that fixes it.

forkineye, 2020-12-02:
This is fixed now. Note however, you need to manually reset your board once programming is complete to see serial output in ESPSFlashTool.

closed 2020-12-02 by forkineye