forkineye/ESPixelStick #246: Reopen: Upload tool for 3.2 not working for Ubuntu 20.04lts #231

extremeskillz, 2020-12-05:
Hi Shelby,
I was able to update that issue I was having in Upload tool for 3.2 not working for Ubuntu 20.04lts #231 
I can't reopen the bug but I figured you can.

Thanks.
Carlos

forkineye, 2020-12-07:
> Hi Shelby,
> I was able to update that issue I was having in Upload tool for 3.2 not working for Ubuntu 20.04lts #231 
> I can't reopen the bug but I figured you can.
> 
> Thanks.
> Carlos
> 

Looks like you don't have python installed? I thought it was by default on Ubuntu. If not, install python 3 and let me know. I may need to add checks for it.

forkineye, 2020-12-07:
Reopened it

closed 2020-12-07 by forkineye

extremeskillz, 2020-12-22:
Python 3 is confirmed and installed on Ubuntu 20.04lts here is the CLI output.
~$ sudo apt install python3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3 is already the newest version (3.8.2-0ubuntu2).
python3 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 27 not upgraded.