forkineye/ESPixelStick #248: Device Setup in ESPixelStick_TRAVIS-20201206132845 build crashes on Save

patdelaney, 2020-12-06:
--------- Instructions - Delete this ---------
Please provide answers directly below each section.
--------- Instructions - Delete to here ---------
**ESPixelStick Firmware Version**
ESPixelStick_TRAVIS-20201206132845

**Hardware Version**
ESP8226 and a relay sheild

**Binary release or compiled yourself?**
ESPixelStick_TRAVIS-20201206132845

**Operating System**

**Access Point**

**Describe the bug**
After uploading the latest ESPixelStick_TRAVIS-20201206132845 build, when I go into Device Setup and try to configure it for a relay, when I click Save changes the unit crashes when it coms back most of the time the settings you made did not stick.

I tried just configuring input then save changes sometimes it works without a crash.  After that then configure the relay, Maybe it will crash and maybe it won't.  Once configured the relay appears to work as designed (single relay) 

I haven't really done and longevity tests to see how it behaves,

patdelaney, 2020-12-06:
Here is the log file
[pixelstick.zip](https://github.com/forkineye/ESPixelStick/files/5649001/pixelstick.zip)

Here is a link to a walkthrough of the test
https://youtu.be/Yv4M8mLSHew

processor83, 2020-12-09:
I had the same problem with last night travis and same thing with today releas. web interface is not stable and will stop responding when i go in device setup. I build the last unify miself and it is super quick and responsive.

MartinMueller2003, 2020-12-14:
Pat and I were using the UI in a teams meeting the other day. Not sure why the manually built images seem more stable than the Travis images.

MartinMueller2003, 2021-01-13:
Not seeing this with the latest RAM reductions.

patdelaney, 2021-02-08:
closing as we have newer Travis builds that are working

closed 2021-02-08 by patdelaney