forkineye/ESPixelStick #251: Entries on Log file don;t seem to correspond with what is configured in device setup

patdelaney, 2020-12-09:
**ESPixelStick Firmware Version**
Unify Branch

**Hardware Version**
Wemos D1mini

**Binary release or compiled yourself?**
Compiled myself

**Describe the bug**
This may be a self inflicted error based on the way that I'm testing, or just my lack of understanding some of the UI elements. But For my relay test (which is running fine) I have configured the device as follows (screenshots to follow)
Starting Universe: 1   Channels per Universe (1) First Universe Chan Offset 1
First Output Mode: Disabled  Second Output Mode: Relay
One relay configured.

I configured FPP to send the same channels to the pixelstick that it's sending to the RGBW strip  that is hanging off a different pixelstick. I'm assuming that the configuration above "listens for universe 1 and channel 1 and "ignores" everything else.

But in the serial log I am seeing:
`16:09:13.429 -> ------ connectWifi:WiFiMgr.cpp:184: End ------
16:09:13.464 -> ** 'Disabled' Initialization for input: '0' **
16:09:13.464 -> ** 'Disabled' Initialization for input: '1' **
16:09:13.464 -> ** 'E1.31' Initialization for input: '0' **
16:09:13.464 -> Configuration file: '/input_config.json' loaded.
16:09:13.464 -> - Web Server started on port 80
16:09:13.464 -> FPPDiscovery subscribed to broadcast
16:09:13.464 -> FPPDiscovery subscribed to multicast: 239.70.80.80
16:09:13.674 -> FPPDiscovery: No SD card
16:09:17.295 -> 
16:09:17.295 -> WiFi Entering State: Connected To AP
16:09:17.295 -> WiFi Connected with DHCP
16:09:17.329 -> WiFi Connected with IP: 192.168.1.110
16:09:17.329 -> E1.31 Multicast Enabled.
16:09:17.329 -> E1.31 Unicast Enabled on port: 5568
16:09:17.329 -> Listening for 8 channels from Universe 1 to 8.
16:09:17.329 -> Sequence Error - expected: 0 actual: 193 universe: 1
16:09:20.912 -> * WS Connect - 1
16:09:20.945 -> File: '/output_config.json' loaded.
16:09:20.945 -> File: '/input_config.json' loaded.
16:09:25.545 -> Sequence Error - expected: 101 actual: 102 universe: 1
16:09:50.378 -> * WS Connect - 2
16:09:50.413 -> File: '/output_config.json' loaded.
16:09:50.413 -> File: '/input_config.json' loaded.
16:09:55.569 -> Sequence Error - expected: 190 actual: 191 universe: 1
16:10:25.628 -> Sequence Error - expected: 23 actual: 24 universe: 1
16:10:55.672 -> Sequence Error - expected: 112 actual: 113 universe: 1
16:11:20.241 -> File: '/output_config.json' loaded.
16:11:20.275 -> File: '/input_config.json' loaded.
16:11:25.742 -> Sequence Error - expected: 201 actual: 202 universe: 1
16:12:25.853 -> Sequence Error - expected: 122 actual: 123 universe: 1
16:12:55.893 -> Sequence Error - expected: 211 actual: 212 universe: 1
16:13:25.960 -> Sequence Error - expected: 44 actual: 45 universe: 1
16:13:55.996 -> Sequence Error - expected: 133 actual: 134 universe: 1
16:14:26.062 -> Sequence Error - expected: 222 actual: 223 universe: 1
16:14:56.111 -> Sequence Error - expected: 55 actual: 56 universe: 1
16:15:26.169 -> Sequence Error - expected: 144 actual: 145 universe: 1
16:15:56.243 -> Sequence Error - expected: 233 actual: 234 universe: 1
16:16:26.309 -> Sequence Error - expected: 66 actual: 67 universe: 1
16:16:56.331 -> Sequence Error - expected: 155 actual: 156 universe: 1
16:17:26.395 -> Sequence Error - expected: 244 actual: 245 universe: 1
16:17:56.453 -> Sequence Error - expected: 77 actual: 78 universe: 1
16:18:26.502 -> Sequence Error - expected: 166 actual: 167 universe: 1
16:18:56.543 -> Sequence Error - expected: -1 actual: 0 universe: 1
16:19:26.613 -> Sequence Error - expected: 88 actual: 89 universe: 1
16:19:56.669 -> Sequence Error - expected: 177 actual: 178 universe: 1
16:20:26.713 -> Sequence Error - expected: 10 actual: 11 universe: 1
16:20:56.744 -> Sequence Error - expected: 99 actual: 100 universe: 1
16:21:26.780 -> Sequence Error - expected: 188 actual: 189 universe: 1
16:21:56.841 -> Sequence Error - expected: 21 actual: 22 universe: 1
16:22:26.893 -> Sequence Error - expected: 110 actual: 111 universe: 1
`
Why would it say:  16:09:17.329 -> Listening for 8 channels from Universe 1 to 8.  
I can't see where I have 8 universes configured.

What do all the sequence errors mean?

MartinMueller2003, 2020-12-14:
It means the WiFi is losing a lot of data. 

The minimum number of channels for the relay is 8. It should be saying U1 to U1

MartinMueller2003, 2020-12-15:
I am seeing the correct number of channels and universe numbers listed.

07:20:17.274 -> WiFi Connecting to MaRtInG as MartinsESP32-2
07:20:17.274 -> ** 'Disabled' Initialization for input: '0' **
07:20:17.274 -> ** 'Disabled' Initialization for input: '1' **
07:20:17.309 -> ** 'E1.31' Initialization for input: '0' **
07:20:17.309 -> Configuration file: '/input_config.json' loaded.
07:20:17.309 -> - Web Server started on port 80
07:20:17.309 -> FPPDiscovery subscribed to broadcast
07:20:17.309 -> FPPDiscovery subscribed to multicast: 239.70.80.80
07:20:17.624 -> FPPDiscovery: No SD card
07:20:18.915 -> 
07:20:18.915 -> WiFi Entering State: Connected To AP
07:20:18.915 -> WiFi Connected with DHCP
07:20:18.915 -> WiFi Connected with IP: 192.168.10.193
07:20:19.019 -> E1.31 Multicast Enabled.
07:20:19.123 -> E1.31 Unicast Enabled on port: 5568
07:20:19.123 -> Listening for 8 channels from Universe 1 to 1.
07:20:20.274 -> * WS Connect - 1

patdelaney, 2021-02-08:
This is now working as expected

closed 2021-02-08 by patdelaney