forkineye/ESPixelStick #252: Slight enhancement tp FPPDDiscovery

patdelaney, 2020-12-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Please make the following code changes to FPPDiscovery.cpp   I have already discussed this with the FPP team and tested it and it works.
I'd like the Build date to the version string that is returned in the fppd discovery. This will populate in the FPP multisync page and will be useful during testing.

In FPP we append the git version to our version number. But that is done on the fly with the build script. Once you move to the new build system we can probably implement that.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

comment out line # 528    
       strcpy (packet.version, version);

add these 3 lines
	strncpy(packet.version, version, 40);
	strncat(packet.version, " ", 40 - strlen(packet.version));
	strncat(packet.version, BUILD_DATE.c_str(), 40 - strlen(packet.version));

**Additional context**
This is what the result would be:
[image]

MartinMueller2003, 2020-12-14:
While the example of the code is good, I intend to leave the line alone and build the desired output string at compile time. Saves code, ram and cpu cycles.

patdelaney, 2020-12-14:
ok

closed 2020-12-14 by patdelaney