forkineye/ESPixelStick #255: xLights upload to set outputs seems to corrupt Device Setup UI

patdelaney, 2020-12-13:
**ESPixelStick Firmware Version**
Latest code from Github (self compiled)
**Hardware Version**
ESP Artstick with SD card

**Binary release or compiled yourself?**
compiled

**Describe the bug**
After uploading the outputs from xLights 
go to Device setup tab and review the configuration  - it looks correct
Go to Second input mode to select Effects
Go to Effects Configuration and the effect dropdown is empty.
[image]

Rebooting the pixelstick doesn't fix the issue.

patdelaney, 2020-12-13:
[image]

patdelaney, 2020-12-13:
After another attempt at uploading from xLights the Relay option populated, but I can't disable it.
[image]

MartinMueller2003, 2020-12-14:
Took a look at this. xLights is sending a config file directly to the ESP using the same web socket commands as the UI. The UI is always in sync with the config on the device since they are released together. The config file being sent by xLights is WAY out of date. I would like to propose that the UI and xLights use different commands to send config data. The ESP will post process and merge the config files from xLights rather than just accept them as it does from the UI. We can do this without making any changes to xLights. I will work on this ASAP.

MartinMueller2003, 2020-12-14:
Update: Once I deleted the controllers in xLights, xLights did a refresh and started sending an updated config. This resolved most of the issues. However, xLights sends invalid values (zero) for group size and zigzag count. Internal code corrects this but did not save the updates in the config files. This means the bad values showed up on the UI. Changes are now on the unify branch to update the config with the corrected values. We could ask the xLights team to always read the config before they send a new set of config changes (aka NEVER cache the config). This would solve the alignment issues when we make updates.

patdelaney, 2020-12-14:
I don;t believe we do that for any other controller. I believe it's a blind
push. But checking with Dan

MartinMueller2003, 2020-12-14:
I can see the read and the write. We just need to make sure xLights always does the read, not only when the controller is first added.

patdelaney, 2020-12-14:
So are you saying when we add a controller in xlights at some point it
queries the controller for the output settings?

MartinMueller2003, 2020-12-14:
Thats what it looks like.

MartinMueller2003, 2021-01-11:
Now that I found and fixed a race condition in the transfer sequences, I retested this again and do not see anything bad happening. I suspect fixing the race condition also fixed this issue since the race condition resulted in corrupted configs as well.

patdelaney, 2021-04-05:
This was fixed in xLights, it was a problem with the WebSocket library.
tested in xlights 2021.11 and 
EsPixelstick
Build Date
Apr 5 2021 - 11:58:32

closed 2021-04-05 by patdelaney