forkineye/ESPixelStick #256: Unify Branch:  Hostname of ESP32 dev board not displayed in Home tab, it shows espressif

patdelaney, 2020-12-13:
**ESPixelStick Firmware Version**
Unify branch latest build

**Hardware Version**
ESP32 Dev Board
**Binary release or compiled yourself?**
Compiled myself

**Describe the bug**
In Wifi Setup I set a hostname.
Reboot ESP32
Check Home page is shows wrong Hostname
It shows espressif instead of esp32-1

[image]

[image]

MartinMueller2003, 2020-12-14:
This is a bug in the ESP32 WiFi core code. 
Added a workaround to force the local DNS cache to become invalidated and thus updates the host name.

patdelaney, 2021-02-08:
This has been fixed and tested
[image]

closed 2021-02-08 by patdelaney