forkineye/ESPixelStick #259: pixel stick 4.0

thedannymullen, 2020-12-17:
Curious, how do I download the piexelstick 4.0 software?  Was looking to try it out.  

Thanks!!

forkineye, 2020-12-17:
It's in the unify branch - https://github.com/forkineye/ESPixelStick/tree/unify.  Also, the releases tagged as "pre-releases" are build off that branch. Note that it's still under development though and there may be issues.

closed 2020-12-17 by forkineye

thedannymullen, 2020-12-17:
Found it!!  Looks like under releases the dev builds are 4.0 .  I missed them somehow the other day when I looked!!  Closing issue!!

MartinMueller2003, 2020-12-17:
Do you plan to use it on Th ESPixelStick hardware or on a different platform? If you have the ESP V3 hardware then you can use a Travis build. If you are using a different platform or want to use the ESP32 version then you need to build it yourself.
You can get the latest source files from "forkineye/ESPixelStick"

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
forkineye/ESPixelStick

Firmware for the ESPixelStick. Contribute to forkineye/ESPixelStick development by creating an account on GitHub.
 |

 |

 |

Follow the readme to build and then upload using the Arduion IDE.
Martin