forkineye/ESPixelStick #261: Device Setup Page Fails To Save With Long MQTT Password

codahq, 2020-12-18:
**ESPixelStick Firmware Version**
3.1, 3.2 (maybe others)

**Hardware Version**
V2 (and JBoards V1)

**Binary release or compiled yourself?**
Binary Release

**Operating System**
Any

**Access Point**
Any

**Describe the bug**
Any

If you enable MQTT and enter in a long password (like the default password created by Home Assistant for example which is 64 characters long) the message is sent to the device over the WS but no response is received.  No error message is returned.  It just fails to save.  Decreasing the length will restore saving and functionality on the page.

forkineye, 2020-12-21:
I'm unable to re-produce this on 3.2. Tried it with 64 and 70 character passwords. Can you get some output from the serial terminal while attempting to set it?

[image]

codahq, 2020-12-22:
*** procS(): Parse Error ***
S2device

codahq, 2020-12-22:
In case this helps.

[image]

codahq, 2020-12-22:
JSON:
S2 {
  "device": { "id": "esps11" },
  "mqtt": {
    "enabled": true,
    "ip": "10.10.10.232",
    "port": "1883",
    "user": "homeassistant",
    "password": "[snipped]",
    "topic": "diy/esps/5ee333",
    "haprefix": "homeassistant",
    "clean": false,
    "hadisco": true
  },
  "e131": {
    "universe": 16,
    "universe_limit": 510,
    "channel_start": 1,
    "channel_count": 600,
    "multicast": false
  },
  "pixel": {
    "type": 0,
    "color": 0,
    "groupSize": 1,
    "zigSize": 0,
    "gammaVal": 2.2,
    "briteVal": 1
  },
  "serial": {
    "type": null,
    "baudrate": null
  }
}

codahq, 2020-12-22:
Sample password:  aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa

closed 2021-09-15 by forkineye