forkineye/ESPixelStick #262: Unify Pixel Count in Device setup screen appears to be wrong based on 3.2 settings

patdelaney, 2020-12-20:
ESPixelstick 3.2
Tested with 8x32 GRB 5v matrix (256 pixels)  (from amazon)
ESPixelstick config:
 
[image]

da_E131 Tester config:  (you must use 510 as the boundary to get this to work)
 
[image]

Pic of Matrix Showing output from above:
 
[image]

Test with Unify branch latest code 
Using the same settings in the da_tester and I have reconfigured ESPixelstick to the same settings as 3.2
 
[image]

Output of Matrix:
 
[image]

forkineye, 2020-12-20:
This has something to do with zig-zag and I'm tracking it down. 0 should be disabled, but it gives the results you're seeing. Set it to 1 and see if it works.

forkineye, 2020-12-20:
Fixed in https://github.com/forkineye/ESPixelStick/commit/64361356360fa16191add3c17484ea005d362bff. The buffer counter was being incremented two steps when zig = 0 and group = 1, causing the output to be cut in half.

MartinMueller2003, 2020-12-20:
Thanks for tracking this down. I was just about to go hunting for it.
Martin

patdelaney, 2020-12-21:
So git pulled the latest code and then recompiled.  I keep getting
WebSocket lost, attempting to connect message. But yet the unit it still
pingable. I have created a recording and posted it here. Note that these
links are unlisted.
https://youtu.be/IvZlPXo0ck4

Steps to reproduce
compile code and upload
Gulp
upload littlefs
configure device
256 nodes
grb color order
brightness .2
Channels per universe must be at 510 to work with this matrix
Save
wait
Change hostname
eventually you get websocket error
clicking on reset button didn;t seem to help
power cycle the stick seems to stabilize it

forkineye, 2020-12-21:
> > So git pulled the latest code and then recompiled. I keep getting WebSocket lost, attempting to connect message. But yet the unit it still pingable. I have created a recording and posted it here. Note that these links are unlisted. https://youtu.be/IvZlPXo0ck4 Steps to reproduce compile code and upload Gulp upload littlefs configure device 256 nodes grb color order brightness .2 Channels per universe must be at 510 to work with this matrix Save wait Change hostname eventually you get websocket error clicking on reset button didn;t seem to help power cycle the stick seems to stabilize it On Sun, Dec 20, 2020 at 2:09 PM Martin Mueller [notifications@github.com](mailto:notifications@github.com) wrote:
> > […](#)
> 
> This along with some crashes while saving is what I've been trying to track down for a while. It's nothing particular to the zig-zag changes.

We'll use this issue for random crashes - https://github.com/forkineye/ESPixelStick/issues/234. Closing this one as zig-zag behavior is what was messing up the pixel count.

MartinMueller2003, 2020-12-21:
WOW. I will try to reproduce later today. It looks like the WiFi settings stored in flash got corrupted. A power cycle will flush them. The ESP code has no access to those other than to force a flush/rebuild cycle.
Martin

MartinMueller2003, 2020-12-21:
Here is an interesting data point. Having the diagnostics screen up grabs over 2KB of ram. and turns the heap into swiss cheese.  Nice feature. Deadly on memory.
Martin
    On Monday, December 21, 2020, 09:22:30 AM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
  WOW. I will try to reproduce later today. It looks like the WiFi settings stored in flash got corrupted. A power cycle will flush them. The ESP code has no access to those other than to force a flush/rebuild cycle.
Martin

MartinMueller2003, 2020-12-22:
While I cannot reproduce the issue Pat found, I have been able to make the ESP8266 version crash pretty quickly. Interesting thing is that I cannot get the ESP35 to crash at all.
   
   - Set up input as E131
   - Output as WS2811 256 pixels
   - sACN tester generating unicast traffic and a fast changing pattern.
   - Bring up the UI
   - Click Device setup then Diagnostics
   - Repeat alternating page as fast as you can
   - ESP8266 will crash within a minute. ESP32 will not have an issue (gave after 3 minutes).

The other thing I noticed is that the Diag screen on the ESP8266 has trouble keeping up with the data stream while the ESP32 Diag screen output is super smooth.
I suspect we are still looking at a low memory issue caused by the UI. At what point do we accept that the UI may be a bit clunky on the ESP8266 but the pixel output is working fine and the ESP is stable as long as you stay off the UI?
Martin
    On Monday, December 21, 2020, 10:50:18 AM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
  Here is an interesting data point. Having the diagnostics screen up grabs over 2KB of ram. and turns the heap into swiss cheese.  Nice feature. Deadly on memory.
Martin
    On Monday, December 21, 2020, 09:22:30 AM EST, Martin Mueller <martinmueller2003@yahoo.com> wrote:  
 
  WOW. I will try to reproduce later today. It looks like the WiFi settings stored in flash got corrupted. A power cycle will flush them. The ESP code has no access to those other than to force a flush/rebuild cycle.
Martin

forkineye, 2020-12-22:
It the sACN tester rate limited? I know some are not and quickly cause
issues. I wouldn't focus on the diagnostics tab too much as there are
enough crashes just accessing the device config tab or trying to save
configuration changes.

forkineye, 2020-12-22:
One delta between the platforms is AsyncTCP vs ESPAsyncTCP.  May be
something going on in there.

MartinMueller2003, 2020-12-22:
On the ESP8266 I got from Shelby: I just sat and clicked from one screen to the next. Alternating between Diags, Device Set up and WiFi Set up as fast as I could for two minutes. Zero crashes. Went the the WiFi screen and clicked save as fast as possible for 30 seconds. zero crashes. Did the same for the device config screen, zero crashes. Turned on E1.31 traffic at a high rate. Alternating between screens, had multiple crashes. Saved as fast as I could, had multiple crashes. 
I think it is an interaction between the E1.31 data path and the UI data path. Let the device sit with the E1.31 traffic turned on (leave the UI alone). No crashes.
E1.31 is unicast. My AP rate limits multicast to the point where most of the E1.31 traffic is lost.
Martin

patdelaney, 2020-12-22:
I configured FPP to mirror the output that goes to 3 snow flakes on the
roof of my house  to 3 snowflakes in the basement during my show. I can see
that they are working when I go down stairs to check, but what I need to do
is capture the serial logs so I can check for crashes and reboots.

I don't really use multicast but I could set it up in the lab and a
separate network if we need it.

Pat

MartinMueller2003, 2020-12-23:
Many hours of E1.31 data with no crashes. 
@Patrick Delaney Just look at the uptime on the UI. That will tell you if the device rebooted.
Martin