forkineye/ESPixelStick #263: IP Fields do not properly validate contents.

MartinMueller2003, 2020-12-23:
**ESPixelStick Firmware Version**
unify - Fetched 2020.12.23

**Hardware Version**
N/A

**Binary release or compiled yourself?**
Self Built

**Operating System**
Win 10

**Access Point**
NA

**Describe the bug**
Brought up the newly modified UI. Much cleaner. One minor problem. If you delete the value in the IP fields, the validation becomes green. It should be red on an empty field.

[image]

forkineye, 2020-12-24:
Should be fixed with https://github.com/forkineye/ESPixelStick/commit/9131893ce9c0f19999dbf00a6bd439fafa98be07.  If / when we decide to enable / disable the save button based upon input validation, we'll need to assign valid default values for the static IP fields.

jnovack, 2020-12-25:
Tasmota sticks with 192.168.4.1.

MartinMueller2003, 2020-12-25:
Setting a default is not a problem. Allowing a blank field is an issue.
Martin

patdelaney, 2020-12-26:
The first time you open the UI after compiling and uploading, and go to the wifi settings, the DHCP is not selected even though that is how it booted and got connected. 
[image]

MartinMueller2003, 2020-12-27:
actually it used dhcp because no address was defined. look at the log. it tells you you have a config error. have not rememberd to fix the dhcp default setting.
martin

patdelaney, 2021-04-08:
closed bogus entries in IP turn it red as expected

closed 2021-04-08 by patdelaney